Blockstream/Jade #1: Missing cbor dependency?

asoltys, 2021-01-05:
I tried running the docker image and got this during the idf.py flash step:

```
-- Building ESP-IDF components for target esp32
CMake Error at /root/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'cbor'.

```

Nathanw1014, 2021-01-05:
I have reproduced the error too. 
`Building ESP-IDF components for target esp32
CMake Error at /root/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'cbor'.
Call Stack (most recent call first):
  /root/esp/esp-idf/tools/cmake/build.cmake:216 (__build_resolve_and_add_req)
  /root/esp/esp-idf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  /root/esp/esp-idf/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:4 (project)`

greenaddress, 2021-01-05:
Hi, this worked for me with the following steps:

```
git clone https://github.com/Blockstream/Jade.git 
cd Jade
git submodule update --init --recursive
docker build . -t jadetest
docker run -v $PWD:/jade jadetest /bin/bash -c "cd /jade && ./format.sh && git diff --exit-code && . /root/esp/esp-idf/export.sh && cp configs/sdkconfig_display.defaults sdkconfig.defaults && idf.py all size-components size"
```

Can you provide the steps that failed for you? thank you

asoltys, 2021-01-05:
I got the error when trying to run with docker-compose as described in the first section of the README:

[image]

asoltys, 2021-01-05:
I was able to build the firmware with your steps and I think I managed to write it to the device with this command:

`docker run -i --device=/dev/ttyUSB0 -v $PWD:/jade jadetest /bin/bash -c "cd jade; /root/.espressif/python_env/idf4.2_py3.7_env/bin/pytho    n ../root/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode     dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x9000 build/partition_table/partition-table.bin 0xe000 build/ota_data_initial.bin 0x100    00 build/jade.bin"`

I got a log of the firmware being written and thought everything went ok but now my device won't turn on. The screen used to show some information about the memory and said "press front button" or something like that but now it's just black/dead.

Here's my terminal output:

```
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a4:cf:12:44:21:10
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash params set to 0x0240
Compressed 18496 bytes to 12997...
Writing at 0x00001000... (100 %)
Wrote 18496 bytes (12997 compressed) at 0x00001000 in 0.3 seconds (effective 471.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 126...
Writing at 0x00009000... (100 %)
Wrote 3072 bytes (126 compressed) at 0x00009000 in 0.0 seconds (effective 5257.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x0000e000 in 0.0 seconds (effective 27963.2 kbit/s)...
Hash of data verified.
Compressed 1118928 bytes to 678773...
Writing at 0x00010000... (2 %)
Writing at 0x00014000... (4 %)
Writing at 0x00018000... (7 %)
Writing at 0x0001c000... (9 %)
Writing at 0x00020000... (11 %)
Writing at 0x00024000... (14 %)
Writing at 0x00028000... (16 %)
Writing at 0x0002c000... (19 %)
Writing at 0x00030000... (21 %)
Writing at 0x00034000... (23 %)
Writing at 0x00038000... (26 %)
Writing at 0x0003c000... (28 %)
Writing at 0x00040000... (30 %)
Writing at 0x00044000... (33 %)
Writing at 0x00048000... (35 %)
Writing at 0x0004c000... (38 %)
Writing at 0x00050000... (40 %)
Writing at 0x00054000... (42 %)
Writing at 0x00058000... (45 %)
Writing at 0x0005c000... (47 %)
Writing at 0x00060000... (50 %)
Writing at 0x00064000... (52 %)
Writing at 0x00068000... (54 %)
Writing at 0x0006c000... (57 %)
Writing at 0x00070000... (59 %)
Writing at 0x00074000... (61 %)
Writing at 0x00078000... (64 %)
Writing at 0x0007c000... (66 %)
Writing at 0x00080000... (69 %)
Writing at 0x00084000... (71 %)
Writing at 0x00088000... (73 %)
Writing at 0x0008c000... (76 %)
Writing at 0x00090000... (78 %)
Writing at 0x00094000... (80 %)
Writing at 0x00098000... (83 %)
Writing at 0x0009c000... (85 %)
Writing at 0x000a0000... (88 %)
Writing at 0x000a4000... (90 %)
Writing at 0x000a8000... (92 %)
Writing at 0x000ac000... (95 %)
Writing at 0x000b0000... (97 %)
Writing at 0x000b4000... (100 %)
Wrote 1118928 bytes (678773 compressed) at 0x00010000 in 16.7 seconds (effective 536.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Nathanw1014, 2021-01-06:
> Hi, this worked for me with the following steps:
>
> ```
> git clone https://github.com/Blockstream/Jade.git
> cd Jade
> git submodule update --init --recursive
> docker build . -t jadetest
> docker run -v [...]

This worked fine until the last line. Using a variant I was able to get idf.py working correctly. Thanks

greenaddress, 2021-01-07:
We updated the docker compose mechanism as it was referring to an outdated version.
Both vanilla docker and docker compose should now work.

Can you confirm it works for you now?

Thanks

asoltys, 2021-01-07:
Nope, looks like I'm still getting the same error about missing cbor dependency.

greenaddress, 2021-01-08:
@asoltys it's probably using the previous docker image you had. Try to remove and rebuild or redownload the docker image please?

asoltys, 2021-01-08:
Ok, I removed the images and the whole Jade directory and re-cloned from github. Getting a new error now following the docker-compose instructions at the `idf.py flash` step:

```
CMake Error at /root/esp/esp-idf/tools/cmake/component.cmake:302 (message):
  Include directory
  '/jade/components/libwally-core/upstream/src/secp256k1/src' is not a
  directory.
Call Stack (most recent call first):
  /root/esp/esp-idf/tools/cmake/component.cmake:470 (__component_add_include_dirs)
  /root/esp/esp-idf/tools/cmake/component.cmake:531 (idf_component_register)
  components/libwally-core/CMakeLists.txt:47 (register_component)

-- Configuring incomplete, errors occurred!
See also "/jade/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

Nathanw1014, 2021-01-08:
@asoltys you need to download the sub-modules of the git repo, you can use `git submodule update --init --recursive` in your existing cloned git repo, or reclone again but add `--recursive`

bavarianledger, 2021-01-09:
I had the same issue like @asoltys.
 
Using
```docker run -v $PWD:/jade jadetest /bin/bash -c "cd /jade && ./format.sh && git diff --exit-code && . /root/esp/esp-idf/export.sh && cp configs/sdkconfig_display.defaults sdkconfig.defaults && idf.py all size-components size"``` instead of ```cp configs/sdkconfig_jade.defaults sdkconfig.defaults``` and ```idf.py flash``` solved it.

When flashing afterwards don't forget to add ```--device=/dev/ttysUSB0``` to the docker command.

But my M5Stack Fire is showing only a white screen after flashing. Any ideas what the issue might be?

JamieDriver, 2022-01-04:
If you want to flash an m5fire you'll need to instead `cp configs/sdkconfig_display_m5fire.defaults.defaults sdkconfig.defaults`.

closed 2022-01-04 by JamieDriver