Blockstream/Jade #9: Jade stuck at "Persisting PIN data..."

Transisto, 2021-01-27:
was setting up the wallet by scanning an existing seed qr code and entered the new PIN twice.

sz3k, 2021-01-30:
Just had a similar issue to this with jade and an android tablet. 

Ultimately, if persisting PIN data takes a while to resolve, the default screen lock/shutdown should be raised or ignored while this process is taking place

max302, 2021-02-27:
I had the same kind of behaviour on iOS. Seeing as the setup phase is quite important, I think it would be good to temporarily disable sleep during the setup phase. 

iOS has a setting for this, not too familiar with android. https://developer.apple.com/documentation/uikit/uiapplication/1623070-isidletimerdisabled

JamieDriver, 2021-06-23:
Some improvements have been made on the ios app to dissuade it from sleeping, or to improve waking/reconnecting.  
Also, in Jade fw version 0.1.25 an 'offline setup' feature was added, where most of the wallet creation/initialisation can be completed before connecting to any companion app.

JamieDriver, 2022-01-04:
Closing - please reopen if you believe there is still an issue here.

closed 2022-01-04 by JamieDriver

hsdredgun, 2024-02-21:
Definitely still an issue with the app Image via USB I have been stuck at Persisting pin data for several hours now...

JamieDriver, 2024-02-22:
The 'Persisting pin data' step should take a few seconds.  If it's taking longer than say 10s then I'd say there's been some issue with networking comms between the app and the blind oracle, or between the app and the Jade.