Blockstream/Jade #13: Not possible tu use Testnet or Liquid

svanstaa, 2021-02-06:
After setting up the HW Wallet with a 24-seed, I tried to change the network in the Android Green Wallet App to Liquid (same with Testnet). Got the response from the app: "The network chosen in the Green App is not the same as the network chosen on the hardware wallet. Please choose same network on both devices".
Also, on the HW device, I did not find an option for changing the network.
So, it is effectively not possible to use Liquid or Testnet at the moment?

JamieDriver, 2021-04-07:
Jade should be able to login/sign on either 'test' networks (eg. testnet, or any regtest or liquid localtest you may run), or 'real' networks (ie. mainnet, liquid) - but not both.
It does not currently allow the use of the same key material on both real and test networks.

JamieDriver, 2022-01-04:
Closing - please reopen if you believe there is still an issue here.  Thanks.

closed 2022-01-04 by JamieDriver

Sjors, 2022-01-07:
Just adding my two cent that this restriction is not ideal. I like to be able to test things on testnet or signet, without having to completely wipe and reconfigure the device. Isn't it sufficient to just enforce the BIP44/49/etc `network` component to be `0` for mainnet and `1` for test networks?

JamieDriver, 2022-12-01:
*Jade* should be able to sign for either 'test' networks (eg. testnet, or any regtest or liquid localtest you may run).
I mean, all Jade really does is hold your wallet seed/keys, and from those provides xpubs or signatures.
It is up to the wallet application to manage which networks are being used.
Jade itself does not manage the derivations being used, it just signs what it's asked to sign with the bip32 path it is given (although with single-sig it may show a warning if the path does not conform to bip44/etc).

Sorry, I can't really comment on using eg. Sparrow.

bitcoinhelp, 2022-12-01:
> Even if I enable the testnet option in the Blockstream Green application, it's not presented as an option for the Jade to login/setup with.

Is your Jade uninitialized? Or is there already a wallet persisted?

You'll only see the testnet option if your Jade hasn't been set up yet. After configuring to use with testnet, you can then use it with Sparrow normally

Example below: 
[image]

strangebit, 2022-12-01:
Thanks for the responses. Ah, I wasn't seeing those options, but I've just figured out it was due to the lack of a scrollbar in the Blockstream Green Desktop app, and my window was _exactly_ the right size to confuse me.

[image]

[image]

JamieDriver, 2022-12-01:
Unlucky!  ;-)

strangebit, 2022-12-01:
Unfortunately I'm still having the same (or similar) error in Sparrow that I was getting yesterday.

    Could not open client or get fingerprint information: JadeError: -32003 - Network type inconsistent with prior usage (Data: None)

It's almost the same error that I was getting yesterday, although it is slightly different now. Now I'm getting this error when I go to 'Connected Hardware Wallet' -> 'Scan' in Sparrow (using USB interface). Previously, I was only getting this error after the scan, and then clicking 'Import Keystore'. Also, the part saying 'Could not open client or get fingerprint information' was not there yesterday. It was just the 'JadeError: -32003' part. See the screenshots below.

I thought that perhaps setting up / initalizing the Jade with a testnet wallet first in Blockstream Green Desktop would solve it, but I couldn't see any way to do this yesterday because of the window size.

I've successfully recieved and sent testnet coins in Blockstream Green Desktop now, so it seems to be working in general. Do we think this is a Sparrow specific problem or maybe HWI or something? If so, I might post an issue on Sparrow GitHub and see.

I'm running Sparrow with `--network testnet`. I have the Jade device on the 'Ready!' screen after using Blockstream Green Desktop to login/unlock the testnet wallet, and if I go to 'Settings' -> 'Xpub Export' on the device I see `m/84'/1'/0'`. I also have the Blockstream Green Desktop completely closed when I'm trying to use the Jade with Sparrow.

Here's screenshots of the two errors. The top one is the one from yeseterday, the bottom one is the one from now.

[image]

[image]

bitcoinhelp, 2022-12-01:
Thanks for the extra detail. I'm assuming you're using 0.1.41? Actually running into the same issue now.. but I have used Jade with Sparrow testnet via USB previously. Going to try to other firmwares and with Specter to see if I can rule it out being a Jade issue. Will get back shortly

strangebit, 2022-12-01:
Yeah, I'm running 0.1.41. Thanks for looking into it! 🤓

bitcoinhelp, 2022-12-01:
@JamieDriver I'm successfully able to use Jade with Sparrow testnet by downgrading to 0.1.39 - I was getting the same issue as described above on 0.1.41

Possible something changed on Jade's end that might be causing this issue on the newer firmware?

cc: @strangebit - workaround seems to be temporarily downgrading to 0.1.39 if it's you're desperate to try sparrow testnet now :)

[image]

strangebit, 2022-12-01:
Interesting. It works after you click 'Import Keystore'' as well, yes? 👍 

How would I go about downgrading the firmware? I don't mind waiting but I'd be interested to know that as well.

bitcoinhelp, 2022-12-01:
No it doesn't. Sorry @JamieDriver 

I got ahead of myself, it scans correctly now - but still the "Network type inconsistent with prior usage error" after Import Keystone

Back to investigating, but downgrading did change the behavior a little

bitcoinhelp, 2022-12-01:
No, sorry. Got ahead of myself. It does not work after selecting "Import Keystone" same network type inconsistent with prior usage.

I tried this with 0.1.41 and 0.1.39 on Sparrow 1.6.5 and 1.7.1 - same issue on all

I then tried with Specter on 0.1.41 and 0.1.39 - no issues importing key using testnet

Perhaps it is worth opening an issue with Sparrow unless Jamie has any extra thoughts

strangebit, 2022-12-01:
All right, thanks for doing those tests. Interesting that it works with Specter no problem. I'll go ask Sparrow about it, and probably link to this discussion from there as well.

bitcoinhelp, 2022-12-01:
Nice. Yeah no problems with HWI over here either.

You're right that removing `--chain test` is normal and will lead to the error - as Jade is now initialized for testnet/regtest etc

strangebit, 2022-12-01:
I think the issue is that `hwi enumerate` or  `hwi --chain test enumerate` returns the result with the error:

    [{"type": "jade", "model": "jade", "path": "COM6", "needs_pin_sent": false, "needs_passphrase_sent": false, "error": "Could not open client or get fingerprint information: JadeError: -32003 - Network type inconsistent with prior usage (Data: None)", "code": -13}]

When Jade is initialized for testnet. Seems to happen regardless of whether `--chain test` is given. I've verified that Sparrow calls `hwi enumerate` when you click 'Scan', but because it returns this result with the error attached, it reasonably assums something went wrong and just shows the error and doesn't let you proceed. This could be considered a Sparrow problem, but I think it's more of a Jade or HWI issue that it's returning the error in the first place?

bitcoinhelp, 2022-12-01:
Super helpful, really appreciate the investigation here.

Will let @JamieDriver chime in/confirm when he's available to see if he can provide some more color

JamieDriver, 2022-12-01:
@strangebit 
Thanks very much for the detailed feedback.

I agree:
```
./hwi.py --debug enumerate
...
{'method': 'auth_user', 'id': '360537', 'params': {'network': 'mainnet', 'epoch': 1669931187}} as cbor of size 63
...
```
and:
```
./hwi.py --chain test --debug enumerate
...
{'method': 'auth_user', 'id': '636000', 'params': {'network': 'mainnet', 'epoch': 1669931365}} as cbor of size 63
...
```
So def looks like an issue with the HWI wrapper.
I will investigate.
(fyi you don't need to worry about 'id:' above - it's a meaningless 'message id' just so we can pair received message replies with the outgoing message request, and detect if we get out of sync somehow)

*But you are right, even if HWI is fixed, the client app would need to pass the `--chain test` option.

fyi @bitcoinhelp

strangebit, 2022-12-01:
Thanks! I'll let Sparrow know as well, or if you want to or @bitcoinhelp wants to then I don't mind.

>fyi you don't need to worry about 'id:' above

Thanks. Trying to be careful if I don't know what something is! LOL better safe than sorry.

JamieDriver, 2022-12-01:
Right,
The issue is the HWI 'enumerate' command completely ignores '--chain' and does not pass it down to each HWW - any wallet client being probed assumes 'Chain.MAIN' as a default.
I guess it was assumed (not unreasonably) that device enumeration was network agnostic - but alas for Jade at least, in order to get an xpub and hence a fingerprint, you have to tell it what network it's for.

With a small change to the HWI code (ie. passing the cmd-line arguments along to the 'enumerate' functions) I can get it to work:
```
./hwi.py --debug enumerate
...
{'method': 'auth_user', 'id': '180162', 'params': {'network': 'mainnet', 'epoch': 1669933615}} as cbor of size 63
...
```
and
```
./hwi.py --chain test --debug enumerate
...
{'method': 'auth_user', 'id': '992665', 'params': {'network': 'testnet', 'epoch': 1669933288}} as cbor of size 63
...
```
I have opened a PR to address this:  https://github.com/bitcoin-core/HWI/pull/651
( Please feel free to lobby for it by commenting on that PR  ;-) )

ofc Sparrow will have to update to that version if/when it is merged, and use the `--chain` option as discussed.