Blockstream/Jade #18: Jade not found during BLE discovery on iOS

btcpuertorico, 2021-02-21:
Jade is not found when trying to pair via BLE on iOS as described on installation guide. Tried two different Jade devices, same behavior.

Transisto, 2021-02-23:
Jade is not yet supported on iOS, the wireless hardware wallet that you see is currently for Ledger X

btcpuertorico, 2021-02-23:
What about this guide? https://help.blockstream.com/hc/en-us/articles/900004040226-Blockstream-Jade-Quickstart-Guide#h_01EYXHZ49AV7HP622MCS65VP1Y

jgriffiths, 2021-02-23:
Jade is supported in https://github.com/Blockstream/green_ios/releases/tag/release_3.5.1 which should reach the app store soon.

max302, 2021-02-27:
I have just setup Jade successfully on iOS using Green 3.5.1, which is the current version on the app store. 

Had to try the setup three times, but it is working. Can OP confirm?

closed 2022-04-11 by JamieDriver