Blockstream/Jade #19: Jade on TTGO T-display?

biterminator, 2021-03-16:
Any help for running Jade wallet on TTGO T-display device?

JamieDriver, 2021-04-21:
Support for this device should be being added in the next fw release.

biterminator, 2021-04-21:
> Support for this device should be being added in the next fw release - the config should soon be available in master branch.

Thank you!

Please post some simple instructions for instalation when released.

JamieDriver, 2021-04-26:
Hi,
We have merged changes/config to support TTGO T-display to master.
This means you should be able to build and flash the firmware as in the README.
The only change you'll need to make, is instead of:
```
cp configs/sdkconfig_jade.defaults sdkconfig.defaults
```
you'll need to:
```
cp configs/sdkconfig_display_ttgo_tdisplay.defaults sdkconfig.defaults
```

Note: on the TTGO T-display board the buttons individually correspond to 'left' and 'right' actions, and both together as 'select'.

JamieDriver, 2021-05-25:
Is this working for you now ?

JamieDriver, 2021-06-23:
Closing - should now be working.

closed 2021-06-23 by JamieDriver