Blockstream/Jade #20: Jade showed the change address as the first one output, why even show the change if it's the same XPUB?

Transisto, 2021-03-24:
This is very confusing for new users.

JamieDriver, 2021-04-07:
Hi,
Jade is not supposed to require the user confirm the change output - if it recognises that it is change.
Can you provide details on how you saw this - in particular what companion app you were using ?
Thanks.

Georgantas, 2021-12-28:
I noticed the change address needs to be confirmed for CSV/Multisig Shield outputs.

JamieDriver, 2021-12-29:
It shouldn't ... but it does rely on the companion application indicating which output is change and providing the key derivation path(s) used to build the redeem script - can you confirm the companion app you were using ?

fyi: If the change info is passed, either a) it is correct and the user does not need to confirm change output, or b) it is incorrect and the call should error.
Only if the change data is NOT passed at all, is the user asked to confirm that output - so this rather suggests something missing in what the companion application is passing to Jade.

Georgantas, 2021-12-30:
Thank you, that makes sense.

I'm asked to confirm the change on the iOS Green app. It must not be passing the metadata. I do not experience this issue using the Green desktop app on Linux.

JamieDriver, 2021-12-31:
Thanks - I will discuss with the ios guys, cheers.

JamieDriver, 2022-03-08:
Hi,
I believe this has been addressed in ios release `3.7.9`' (of the Green wallet app).
Please reopen this item if this is not the case.
Many thanks.

closed 2022-03-08 by JamieDriver

Georgantas, 2022-03-08:
Yep, can confirm it's resolved now. Thank you!