Blockstream/Jade #21: Compatibility of Jade with Liquid

stepansnigirev, 2021-09-13:
It looks like Jade only can sign transactions that were blinded by Jade itself - values and assets are displayed correctly only if I pass commitments that contain `hmac` field received from Jade during blinding.

If I pass `None` for all commitments it displays garbage on screen and then throws an error:
```
JadeError: -32602 - Failed to verify asset_generator from commitments data
```

If I pass correct commitments but without `hmac` field - it throws a different error:
```
JadeError: -32602 - Failed to extract trusted commitments from parameters
```

This means that:
1. It's not possible to use Jade in multisig unless Jade is creating the transactions (so 2 Jades will never be able to sign multisig transaction)
2. It's not possible to use Jade for transactions containing outputs that are blinded by someone else (for example swaps, coinjoins or payjoins)

I suggest that you allow commitments that do not contain `hmac` field because HW still can verify correctness of asset commitment and value commitment just by running `secp256k1_generator_generate_blinded` etc. If you already have this functionality in HW why do you force to use blinding factors from HW?

JamieDriver, 2021-09-20:
Acknowledged.  Will add to backlog.

JamieDriver, 2022-04-26:
@stepansnigirev [Some of] this should be included in the next Jade fw release - `0.1.34`.
ie.
sign-liquid-tx:
- We do not read or verify any 'hmac' field in the commitments (if present, it is ignored [like any other unexpected key])
- We do not enforce that the expected 'deterministic blinders' are used - any blinding factors are accepted
- commitment-data passed into sign-liquid-tx must to be correct/consistent (ie. we check the given commitments, given the bf's and unblinded values)
- The given commitments are then copied into the tx, and signed (as now)

get-commitments:
- 'get-commitments' remains as it is, but does not compute or return an 'hmac' field.  ie. it still computes deterministic blinders as per Ledger (assuming the correct 'hashprevouts' and 'index' is passed)
- but the 'get-commitments' step is now optional - the blinding could be done by 3rd party, and the blinders need not be deterministic (or could be deterministic by some other scheme of which Jade is unaware).

Outstanding atm:
- optionally include a master blinding key in the multisig descriptor  -- ie. so jade can be called to do blinding (commitments, blinding factors, blinding key, blinding nonce) of behalf of the multisig wallet (rather than always using it's own signer slip77 key)

- Handling fields with unknown blinders - not planned for 0.1.34, but yes, accept this needs to be handled.

JamieDriver, 2022-06-08:
FYI `0.1.34` is now released, and should support generic liquid multisig and external blinding.

* Does not yet support outputs where the blinders are not known
* Does not yet verify or sign the rangeproof