Blockstream/Jade #23: Jade on Liquid - sign with sighash ALL | RANGEPROOF

stepansnigirev, 2021-09-13:
Signatures returned from Jade are not commiting to rangeproofs. This can cause lock of funds.

Malicious software wallet can generate invalid rangeproof in change output of the transaction using an attacker's blinding key. Then the user will not be able to unblind change output and will not be able to spend this money.

It's not a loss of funds, but certainly lock of funds and an opportunity for ransomware attacks.

JamieDriver, 2021-09-20:
Acknowledged.

stepansnigirev, 2022-02-22:
When / if you will be implementing that, would it be possible to also support external master blinding keys for multisig wallets?
How I see it:
- if the wallet is single-sig and the key belongs to the device - verify that correct blinding key is used (slip77 from seed) - check you can rewind the rangeproof, if all good - it's a change output
- if multisig is used - allow master blinding key for the wallet to be set externally (when multisig wallet is registered, for example). And use this master key to check you can rewind the proof in multisig change outputs

JamieDriver, 2022-02-23:
Yes, agree with the above or something v similar.  Hopefully we'll be looking at this soon.  I'll keep you informed.