Blockstream/Jade #29: Interesting article: https://shiftcrypto.ch/blog/the-pitfalls-of-multisig-when-using-hardware-wallets/ Wonder if multisig with Jade is vulnerable to some of these attacks.

Georgantas, 2021-12-28:
Interesting article: https://shiftcrypto.ch/blog/the-pitfalls-of-multisig-when-using-hardware-wallets/ Wonder if multisig with Jade is vulnerable to some of these attacks.

_Originally posted by @Georgantas in https://github.com/Blockstream/Jade/issues/22#issuecomment-1001681184_

Georgantas, 2021-12-28:
Actually, based on [this](https://github.com/Blockstream/Jade/issues/20#issuecomment-814934215) comment, it seems SingleSig/regular MultiSig change does not need to be confirmed by the user (I was checking MultiSig Shield change).

In that case, can a malicious computer omit the change address completely causing the user to burn the change they should have gotten?

JamieDriver, 2021-12-29:
At the moment, no - it does not verify or mandate the presence of a change output.

Georgantas, 2021-12-30:
Makes sense, I forgot the Jade asks the user to confirm the fee 🤦‍♂️ Thank you :)

closed 2021-12-30 by Georgantas