Blockstream/Jade #30: Add a check on multisig paths

Georgantas, 2021-12-30:
> Thanks. So with Jade we opted to be quite strict in this regard - in order to use multisig the user first needs to register the setup on Jade - ie. as a separate registration step the user inspects/confirms:
> 
>     * all co-signer xpubs (the hw checks that at least one belongs the Jade)
> 
>     * threshold
> 
>     * script type
> 
>     * (and now) whether bip67 key sorting applies
> 
> 
> Once that has been confirmed, future requests to get receive addresses or change addresses will use the registered setup - the only additional data passed per request is the bip32 path suffixes. The full key derivation paths are validated against bip45/48/87, and that the final element is within a reasonable range. If this validation fails, a warning is displayed and the user must explicitly check/confirm the path.
> 
> We believe this should cover the known attack scenarios - but obviously are happy to hear if you think otherwise! ;-)

I think I thought of a scenario.

I noticed that to get an address, the computer sends the following to the Jade:
```
INFO:jade:Sending: {'method': 'get_receive_address', 'id': '468131', 'params': {'network': 'mainnet', 'paths': [[0, 2], [0, 2], [0, 2]], 'multisig_name': 'hwi'}} as cbor of size 108
```

However, I see that the Jade accepts paths that don't match like:
```
INFO:jade:Sending: {'method': 'get_receive_address', 'id': '865589', 'params': {'network': 'mainnet', 'paths': [[0, 1], [0, 2], [0, 3]], 'multisig_name': 'hwi'}} as cbor of size 108
```
which I think is non-standard in most companion wallets.

Could an attacker pass something like `[[0, 2333], [0,3999], [0,4383]]` and perform a type of ransomware attack on `get_address` or `sign_tx`, where they will only reveal the paths in return for some BTC?

I was happy to see there's actually some validation on paths here: https://github.com/Blockstream/Jade/blob/34a1814bbde5486e08600de701619fb7779b5455/main/wallet.c#L343 , but I don't think this case is covered. I think this wouldn't be a problem for 2 of 3 multisig, as the victim could easily brute force the paths because there's a check on `MAX_PATH_PTR` of 10000, so 10000^3=10^12 combinations to try (for comparison, bitcoin network hashrate is in the order of 10^20 Hashes/second), but this might be a problem for larger multisigs (I see up to 8 signers are supported)? Should we be checking that the paths are all equal to each other for multisig wallets?

_Originally posted by @Georgantas in https://github.com/Blockstream/Jade/issues/22#issuecomment-1002841897_

JamieDriver, 2021-12-31:
Yes, it might be a good idea.  
I initially decided against baking in the 'all paths (suffixes) must be the same' (hence passing one path suffix per signer, rather than just one common suffix) but I agree the combination of many signers may be difficult to brute-force.  
Maybe display a warning if path suffixes are not identical ?

Georgantas, 2021-12-31:
Yeah, I think a warning is a good idea to still allow users the flexibility.

JamieDriver, 2022-01-04:
I'll add this into the next fw version.  Thanks.

JamieDriver, 2022-03-08:
Warning as described added to fw version `0.1.33`.
Many thanks for the review/suggestion.

closed 2022-03-08 by JamieDriver