Blockstream/Jade #31: Feature Request: Airgap functionality

evliu, 2022-01-06:
Hi, I couldn't find any issues mentioning this, but am wondering if there is a roadmap for this. Bought multiple Jades based on this Blockstream blogpost:
https://blog.blockstream.com/en-secure-your-bitcoin-and-liquid-assets-with-blockstream-jade/
specifically this paragraph:

> **Where We’re Going, We Don’t Need Cables**
_Blockstream Jade also has a few “dormant” features that we’ll be activating through software updates over the course of 2021. Many bitcoiners (especially us!) are looking forward to the ability to properly airgap their hardware wallets. So we’ve equipped Blockstream Jade with an on-board camera._
>
> _Once activated, the camera will enable you to complete all incoming and outgoing transactions entirely through QR codes, ensuring that you don’t even need to connect your Blockstream Jade to your device through a cable._

Since 2021 is over, just wanted to see how this is going.

Thank you guys!

JamieDriver, 2022-11-16:
Initial PSBT signing and receive address validation (singlesig) released in `0.1.41`.  
Apologies for the delay!

In upcoming releases we hope to improve multisig support (signing currently works, but address validation (including change addresses when signing) does not work for multisig atm), and provide a way to persist the wallet mnemonic/seed on device and decrypt/unlock it using the blind pinserver via QR codes  (ie. same as when using usb/ble), thus saving you having to have a QR code of your wallet/seed lying about.

We also hope to provide a site showing a QR code of the current epoch time, to initialise Jade for TOTP use when airgapped (also unsupported atm).

evliu, 2022-11-16:
Awesome! I've been experimenting with it and working w/ Rich on feedback. I proposed creating an app for the blind pin server (was looking at the green iOS repo and seeing if I could simplify it to be an app only with the jade swift lib, but didn't have time). Created a request on this ticket https://github.com/Blockstream/green_ios/issues/45

JamieDriver, 2023-02-15:
Further enhancements, mainly to multisig support, in fw `v0.1.42`.
'Sign message' should be also be supported in fw `v0.1.44`

JamieDriver, 2023-02-27:
Unlock with PIN (ie. the pinserver protocol) using QR codes implemented in `v0.1.45`.  
ie. 'airgapped' users can now opt to either (depending on their choice of tradeoffs - eg. whether they prefer to have a QR code of the wallet to hand vs having encrypted wallet data persisted in Jade flash and some/partial encryption-key-data stored on the pinserver) :

a) scan a QR of their seed phrase ever time they want to use Jade (and make a CompactSeedQR when initially entering the phrase and recording on [these templates](https://help.blockstream.com/hc/en-us/article_attachments/10890522035225/CompactSeedQRTemplate.pdf)).

b) Use the blind pinserver to assist in encrypting the wallet seed, which can then be safely persisted in the Jade flash.  Pinserver interaction is then needed to decrypt those keys, using eg [this web application](http://blockstream.com/pinqr).
(Note: more advanced users can run their own pinserver, and are not constrained to using the blockstream-provided instance).

NOTE: in either case the user can use bip39 passphrases to further protect their wallet.

bitcoinprecept, 2023-05-01:
This request seems to be delivered. Leaving open to track expanded functionality?

JamieDriver, 2023-05-12:
You're probably right, we can probably close this.  
As you say there may be ongoing improvements/extensions to what we can do with QRs, but that's (hopefully!) no different to any other existing feature.  
Closing as delivered.  Thanks.