Blockstream/Jade #33: Derivation path for Jade with Passphrase

rc-fire, 2022-02-07:
I have a wallet (with passphrase) created on a Jade hardware wallet and I would like to recover the wallet in Electrum. After all it should be BIP39 compatible.

I entered the seed, checked the BIP39 marks, typed in my password as "additional words" (which works fine for ledger or trezor for example), and then chose the right derivation path (which should be p2sh-segwit, since my addresses all start with 3), but I dont get the same addresses as in the green software.

I tried changing the account number of the derivation path (so m/49'/0'/1' instead of m/49'/0'/0') but it is also not working.

Does anyone know why this is incompatible and what could be done to fix it?

JamieDriver, 2022-02-18:
We tested with passphrase with Jade against Trezor - same mnemonic and same passphrase - and they arrived at the same wallet.  
Let me know if you don't agree and we'll do some more investigation.
Yes, certainly we just follow bip39.
Thanks.

bitcoinhelp, 2022-09-21:
@rc-fire it's sounds possible that you were using Jade in a Multisig Shield setup (based on your post date). If you were using Jade with Green, singlesig login with HWWs was not available until May 2022. 

This would explain why you did not see the same wallet in Electrum

closed 2023-02-27 by JamieDriver