Blockstream/Jade #35: CSV change not being detected using iOS companion app

Georgantas, 2022-03-13:
It seems the issue described [here](https://github.com/Blockstream/Jade/issues/20#issuecomment-1062076826_) has resurfaced.

FW version: 0.1.33
iOS App Version: 3.7.9

Georgantas, 2022-03-13:
Maybe this could be fixed on the Jade's side by having it check some default change paths (like the input path + 1, and the common csv_block times [25920, 51840, 64435])?

Another possible solution is maybe to register CSV accounts on the Jade similar to the multisig implementation? In this approach, if an account is registered with an xPub that isn't Blockstream's, a warning could be displayed during registration.

JamieDriver, 2022-03-14:
Cheers - will pass this on.
fyi jade should display a warning if the csv value is not in the expected list.

JamieDriver, 2022-04-26:
Can you try in iOS App Version: 3.8.0 pls.  Many thanks.

Georgantas, 2022-04-28:
Yep, all good now.

Thank you.

closed 2022-04-28 by Georgantas