Blockstream/Jade #36: Register CSV accounts on the Jade

Georgantas, 2022-03-13:
It would be nice to be able to register CSV accounts on the Jade similar to the multisig implementation. If an account is registered with an xPub that isn't Blockstream's, a warning could be displayed during registration.

I think this would make Jade's interface more consistent as well. For sending a txn, the `change` parameter would be the `path` and the `account` (no more [`block`](https://github.com/Blockstream/Jade/blob/master/test_data/txn_2of2csv_change.json#L5) parameter). I think it would also be more secure because an attacker might pass a non-standard `block` parameter (not sure if this would trigger a warning currently).

JamieDriver, 2022-03-14:
fyi jade should display a warning if the csv value is not in the expected list.

georgantas, 2024-09-11:
I think this should be possible with miniscript descriptors now. I will resolve the issue.

closed 2024-09-11 by georgantas