Blockstream/Jade #37: Error Upgrading Firmware 0.1.33

cjrequena, 2022-03-14:
When upgrading the firmware to version 0.1.33 an error appears "ERROR_DECOMPRESS"

JamieDriver, 2022-03-14:
I suspect this error suggests that the data received by the jade unit was corrupted - I think it's pretty unlikely that it was corrupted during the download from the fw server - much more likely that it gets corrupted during upload to Jade over BLE/serial.
Can you give details and I'll see if I can reproduce locally ?  

- Does it error at the same '%' each time, or does it appear randomly ?
- Do you have a jade with a true wheel, or a rocker/jog wheel ?
- Were you using BLE enabled (ie the default) firmware or is it the 'no-radio' variant ?  (ie. can you enable BLE on the unit [via the Bluetooth screen]) ?
- Were you doing the update over serial or BLE ?
- Which O/S app was being used (android, ios, desktop) ?

Many thanks.

cjrequena, 2022-03-14:
- Does it error at the same '%' each time, or does it appear randomly ?
Yes, for radio firmware is always at 3% and for no-radio firmware is always at 41%

- Do you have a jade with a true wheel, or a rocker/jog wheel ?
It is a rocker/jog wheel, I bought it last week
   
- Were you using BLE enabled (ie the default) firmware or is it the 'no-radio' variant ? (ie. can you enable BLE on the unit [via the Bluetooth screen]) ?
I am using the BLE enabled by default, it can be connected via bluetooth but I am using a cable. It is the  first time I am trying to upgrade it, by default came with 0.1.32 radio version
   
- Were you doing the update over serial or BLE ?  
Over serial
   
- Which O/S app was being used (android, ios, desktop) ?
MacOS Monterrey, desktop

JamieDriver, 2022-03-14:
Thanks for the details - I'll try to do same locally.

JamieDriver, 2022-03-14:
So I've just OTA'd a rocker-wheel hw unit twice with no issue.  I am on linux though, not macos (but that shouldn't really make any difference).  I asked a colleague who uses mac-os and he said it was fine for him also - although we did discuss the annoying failures that can be introduced by a dodgy usb cable (although I'd expect that to be 'random', rather than always at the same %).  
But just as an idea - do you have other cables you can try ?

cjrequena, 2022-03-14:
I am using the cable that came with Jade, tomorrow I will try with another one and also via bluetooth

cjrequena, 2022-03-15:
I was able to upgrade the firmware via bluetooth using a mobile phone. I tried with another cable but I got the same error, via bluetooth works fine.

Many thanks.

JamieDriver, 2022-03-15:
Great to hear - sorry you were having issues.  
We are working at the moment on making the size of the update binary which needs to be transmitted to the hw much smaller, which will hopefully mean OTA is faster and more reliable in future.

JamieDriver, 2022-03-15:
Sorry, do you know the desktop app version that had issues, and the (ios?) phone app version that worked ?
(Just so we can check we're testing the exact same versions you had the problem with - many thanks)

JamieDriver, 2022-03-15:
Thanks for the info - I meant also the version of the Green app - eg. 1.0.5 or 1.0.6 for desktop app, 3.7.8 or 3.7.9 for the ios app ?
Sorry to pester, but this sort of feedback really is useful.  Thanks.

cjrequena, 2022-03-15:
- Green v1.0.6 for MacOS Monterrey 12.2.1
- Green v3.7.9 for Ios 15.3.1

JamieDriver, 2022-03-15:
Many thanks, much appreciated.

JamieDriver, 2022-03-21:
Just fyi: we've had a few users having issues on MacOS 12.2.1 so maybe related to OS, both OTA and signing tx - investigating.

JJDsource, 2022-05-24:
Just purchased a Jade and just installed Blockstream Green.
Initial setup 

Getting the same error Mac 12.3.1.

I'll try on bluetooth or android later.

JJDsource, 2022-05-24:
> Just purchased a Jade and just installed Blockstream Green. Initial setup
> 
> Getting the same error Mac 12.3.1.
> 
> I'll try on bluetooth or android later.

Windows PC. Firmware upgrade zero issues first try.

JamieDriver, 2022-05-24:
So it appears that some versions of MacOS do not have the best/latest driver for the usb/serial chip (WCH CH9102F), which can cause problems when passing a lot of data over serial.  
There should be newer/updated/proprietary drivers available which resolve these issues, but we're not really in a position to test/verify/recommend closed-source third party libs which are not already included as part of the O/S.

We are trying to work around this by having the desktop app send the OTA data in smaller chunks on MacOS, which appears to alleviate most of the problems.
This change should be included in the next release of the [green_qt desktop app](https://github.com/Blockstream/green_qt).

I will update here when available.

JamieDriver, 2022-06-08:
[green_qt release 1.1.0](https://github.com/Blockstream/green_qt/releases/tag/release_1.1.0) is out, and should contain the 'smaller upload chunks' fix/workaround (although the best experience would be to use an improved driver).

cjrequena, 2022-06-18:
Hi, I just updated the firmware on a mac and works fine. Using green 1.1.1.

JamieDriver, 2022-09-01:
Was mac-specific app issue, so closing here.  
(NOTE: the apps should now be able to use the much smaller 'delta' fw upload, which should also reduce size/time/errors).

closed 2022-09-01 by JamieDriver