Blockstream/Jade #39: [Feature] Allow restore via scanning of seeds with 4 letters per word.

Transisto, 2022-04-12:
This is what coldcard BIP-85 QR export use.

It makes sense since there's already error correction in the QR and that too many letters create too dense QR to scan on lower res cameras.

JamieDriver, 2022-05-06:
Nice suggestion, thanks!  
We will extend qr mnemonic scanning to not necessarily require the entire mnemonic / full words, but a space separated list of word prefixes - so long as each given prefix is unambiguous, it will be expanded into the full mnemonic word.

eg: qr code of string:
`fish inne face ging orch perm usef meth fenc kidn chuc part favo suns draw limb scie cran oval lett slot invi sadn bana`
should work and get the normal Jade test wallet (as used in the unit tests).
(NOTE: Not restricted to 4-char prefixes - any prefix length is fine so long as it is unambiguous/unique.)

JamieDriver, 2022-09-01:
NOTE: `0.1.35` also supports SeedSigner's 'SeedQR' format.  
SeedSigner's 'CompactSeedQR' will be supported in `0.1.38`.
Importing BC-UR Bip39 format is also on the roadmap ...

closed 2022-09-01 by JamieDriver