Blockstream/Jade #44: Add support for importing seeds through SeedSigner SeedQR codes

eliasnaur, 2022-08-15:
From empirical experiments, the Jade device does not recognize the [SeedQR](https://github.com/SeedSigner/seedsigner/blob/dev/docs/seed_qr/README.md) (and CompactSeedQR?) formats championed by seedsigner.com. Can support be added to Jade?

JamieDriver, 2022-08-15:
The upcoming release `0.1.35` of Jade will support importing SeedQR mnemonics (it should already be in the master branch).  

There were concerns about CompactSeedQR as it lacks the bip39 checksum values that surely have value when transcribing data by hand, and support has not been added for this format at this time.  ofc it could be added in a future release if there is strong demand.

closed 2022-08-17 by eliasnaur

JamieDriver, 2022-08-30:
FYI Support for scanning CompactSeedQR has been merged to the master branch, and will be included in a future release, tentatively `0.1.38`.

JamieDriver, 2022-09-21:
FYI `0.1.38` will also be able to display a CompactSeedQR via multiple 'zoomed-in' images - much like SeedSigner itself, and we should be making paper templates available to facilitate making a qr-code recovery backup.