Blockstream/Jade #45: Allow Web Apps to integrate Jade

tiero, 2022-08-16:
Is there a plan to support Web USB and/or Web Bluetooth protocols?

## What

Allow browser apps to interact with Jade for doing various lifecycle activities, such get new addresses or signing PSBT/PSET?

## Why

Especially on Liquid, a vibrant ecosystem of web apps that offers advanced use cases is growing 

- [BitMatrix](https://beta.bitmatrix.app/)
- [Fuji Money](https://fuji.money/)
- [TDEX](https://dex.vulpem.com/)
- [SideSwap](https://sideswap.io/swap-market/?product=USDt)

The only solution is using Marina Wallet, a browser extension at the moment, Although this makes a very nice UX on for desktop users, it degrades the security trade-offs.

## How

I know Jade is tightly coupled with GDK, but IMHO this may not be the best API and maybe will put too much pressure on the GDK team which could slow down this. What if we could offer a greenfield API, tailored to web/mobile apps developers?

Marina Web Provider could be a starting point for the discussion

https://docs.vulpem.com/marina/api

JamieDriver, 2022-08-17:
FYI:
Jade is not coupled at all with GDK - the Green apps that interface with Jade use GDK, but Jade itself does not.
Jade uses CBOR messages as an interface, so anything that can send/receive CBOR can communicate with Jade - most easily over usb/serial.
eg. HWI, Electrum, Sparrow, Spectre - these support Jade and are not GDK-based apps.

I guess a webapp would need to be running in a browser than supports WebSerial - eg. Chrome (Edge?)

re: PSBT - atm Jade does not talk PSBT, although this is something we want to support soon.
(We also want to support more comms via QR codes, which may or may not be relevant to you.)

tiero, 2022-08-17:
Thanks for the prompt response, yes after my message I read more in the code, you're right actually any interface can be made on top.

Re: WebSerial: Yes I guess chrome, edge should support it, maybe also firefox? Metamask allow it somehow

As reference I am looking to something Ledger offers https://developers.ledger.com/docs/transport/web-hid-usb/

tiero, 2022-12-01:
Hey, @JamieDriver do you have any updates on the possibility of web apps interacting with an attached Jade over USB/Bluetooth?

Just to mention our use case and needs, I guess all boils down to have the Jade the following two APIs:

- `get_public_key()` This allows the app to create customs tapscripts, such as covenant, and use the Jade as a future signer to satisfy CHECKSIG operation 
- `sign_schnorr(public_key, sighash)` this will allow passing the custom-made sighash built by the app to be signed using the previously fetched public key. 

I recognize there are some security concerns such low-level API, and tricking user to sign tx that spends all his funds to the attacker, but I guess can be mitigated by passing the whole preimage hash, so that can be matched and shown to user?

The alternative is to use PSET, but I guess that can take much more time to be ready?

JamieDriver, 2022-12-02:
Web apps should be able (should always have been able) to connect to Jade over USB, provided the browser supports WebSerial.
FYI we have an work-in-progress js web-app that connects to Jade in order to update the firmware - that isn't want you want to do I know, but it's the same in that it connects and then passes cbor messages to the Jade and gets the responses.
It should be available soon, and you'll be able to look at the javascript.

* `get_public_key` - yes, jade supports a 'get_xpub' message which takes a derivation path
* `sign_schnorr` - Jade has 'sign tx' and 'sign psbt' for BTC, and a 'sign_liquid_tx' for liquid (`sign_pset` may come at some point).  You pass in the path (rather than the public key - signatures are made with the private key so you need the path to derive that) and the tx (and maybe the prior tx's also).   atm jade only signs SIGHASH_ALL, although that will be extended in the future to support more complex tx types (swaps, loans, coinjoin, payjoin, etc).

tiero, 2022-12-02:
Thanks, Jamie. 

Having a documentation/dev portal even before releasing the full APIs, would be great to pull in external app developers and help in shaping the API spec together with the end users in mind. Happy to help on that

I imagined supporting TR inputs may take a lot of time, that's why maybe a lower-level method can be more flexible at this stage? Unfortunately, to use the new Liquid opcodes you must use them in the script path spend of TR output, which would greatly help smart contract developer to give a secure way to interact with them.

Thanks again, keep up the great work

CoderBoogs, 2023-02-02:
Expose the API!

JamieDriver, 2023-02-02:
^ hi, sorry what do you mean by this ?  What API are you referring to ?

Jade's 'API' is a messaging protocol using cbor messages, as documented [here](https://github.com/Blockstream/Jade/blob/master/docs/index.rst)  
There is no other 'API' for Jade.

CoderBoogs, 2023-02-10:
I saw this in a telegram chat I think it was about allowing offline trades straight from Jade

CoderBoogs, 2023-02-10:
For the HWW integration, still pending Blockstream/Jade exposing API to handle signing taproot transactions.

Can’t say any ETA, I guess is not even in their backlog yet. If all of us show more interest in this feature they may change priorities. Adding a comment in this GitHub issue or in the Jade Telegram group can help.
https://github.com/Blockstream/Jade/issues/45

JamieDriver, 2023-02-10:
Ah the taproot/schnorr sigining ?   Right - that is definitely something we need to add - agree.

JamieDriver, 2023-02-27:
re: https://github.com/Blockstream/Jade/issues/45#issuecomment-1334906955

Not schnorr yet (sorry) but fyi we have a web page which interacts with Jade using webserial (in this case just to update the firmware - but this process uses calls in the cbor api, same as any other interaction would.)
https://jadefw.blockstream.com/upgrade/fwupgrade.html

Perhaps requires Chrome ?  Not sure which/if other browsers support webserial.

JamieDriver, 2025-03-19:
fyi: as of release `1.0.34` jade supports schnorr signatures and bip86 singlesig taproot.