Blockstream/Jade #52: it is possible to flash inside a ESP32 Basic Core

vincenzopalazzo, 2022-12-28:
I ran the following commands

```
git clone --recursive https://github.com/Blockstream/Jade.git $HOME/jade
cd $HOME/jade
cp configs/sdkconfig_jade.defaults sdkconfig.defaults
idf.py flash monitor
```

But I'm not able to boot after the fash my ESP32 Basic Core https://shop.m5stack.com/products/esp32-basic-core-iot-development-kit-v2-6

I keep receiving the following info

```
Writing at 0x00124122... (79 %)
Writing at 0x0012bbe2... (81 %)
Writing at 0x001362da... (83 %)
Writing at 0x0013dd30... (85 %)
Writing at 0x00143670... (87 %)
Writing at 0x001490c3... (88 %)
Writing at 0x001509b0... (90 %)
Writing at 0x001565cb... (92 %)
Writing at 0x0015bda3... (94 %)
Writing at 0x0016173f... (96 %)
Writing at 0x00166d17... (98 %)
Writing at 0x0016c69f... (100 %)
Wrote 1437664 bytes (875724 compressed) at 0x00010000 in 21.5 seconds (effective 534.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 126...
Writing at 0x00009000... (100 %)
Wrote 3072 bytes (126 compressed) at 0x00009000 in 0.1 seconds (effective 374.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x0000e000 in 0.1 seconds (effective 492.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Serial port /dev/ttyUSB3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.......
Detecting chip type... ESP32
Running idf_monitor in directory /home/vincent/Github/Jade
Executing "/home/vincent/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/vincent/Github/Jade/tmp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB3 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 1 /home/vincent/Github/Jade/build/jade.elf -m '/home/vincent/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/home/vincent/Github/Jade/tmp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0138,len:1496
load:0x40078000,len:14572
ho 0 tail 12 room 4
load:0x40080400,len:2956
0x40080400: _init at ??:?

entry 0x400805d4
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0138,len:1496
load:0x40078000,len:14572
ho 0 tail 12 room 4
load:0x40080400,len:2956
0x40080400: _init at ??:?
```

I'm doing something wrong? or it is not possible to use the firmware inside the https://shop.m5stack.com/products/esp32-basic-core-iot-development-kit-v2-6

odudex, 2023-01-02:
You're using the Jade hardware config file, use M5 black instead. Change third command to:
`cp configs/sdkconfig_display_m5blackgray.defaults sdkconfig.defaults`

vincenzopalazzo, 2023-01-04:
Mh, yeah I tried also this during my install experience but I keep receiving the following error 

```
Running idf_monitor in directory /home/vincent/Github/Jade
Executing "/home/vincent/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/vincent/Github/Jade/tmp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB3 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 3 /home/vincent/Github/Jade/build/jade.elf -m '/home/vincent/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/home/vincent/Github/Jade/tmp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0138,len:1496
load:0x40078000,len:14572
ho 0 tail 12 room 4
load:0x40080400,len:2956
0x40080400: _init at ??:?
```

JamieDriver, 2023-01-05:
After you `cp configs/sdkconfig_display_m5blackgray.defaults sdkconfig.defaults`, you may need to also:
```
rm sdkconfig
idf.py fullclean
```
just to make sure any files created with the inappropriate config are cleaned.

vincenzopalazzo, 2023-01-05:
Oh, looks like that I ran the opposite commands

```
idf.py fullclean
rm sdkconfig
```

Now worked, thanks

JamieDriver, 2023-01-05:
I just managed to find my M5 'basic' (black base/body)  which I think is similar and that worked with latest master - so cool - should all be working then.
Many Thanks, closing this item.

closed 2023-01-05 by JamieDriver