Blockstream/Jade #53: Bug in xpub export

SatsCzar, 2023-01-16:
I found a problem when exporting the xpub from the Wrapped Segwit wallet (BIP 49), when importing into the Blue Wallet, it imports addresses that are Native Segwit, but Jade checks these addresses and does not identify them as belonging to my seed.
I'm using the latest version of the firewall, I'm investigating further to find out if it's a problem with Jade or BlueWallet.

What other android apps could I use to import a Watch-Only?

I should send more information about the bug and how to replicate it soon.

JamieDriver, 2023-01-16:
This could be a facet of this item we opened on BlueWallet about the wrapped segwit import:
https://github.com/BlueWallet/BlueWallet/issues/5136  ?

odudex, 2023-01-16:
Try Nunchuk on Android

SatsCzar, 2023-01-16:
Makes sense, is a problem with BlueWallet.
I exported the xpub and put in Blockonomics website and works fine!

closed 2023-01-16 by SatsCzar