Blockstream/Jade #56: Please add brightness +/- when scanning QR code

danskich, 2023-03-18:
Please add the ability to adjust brightness when scanning QR code. Jade won't scan 99% of the time because of the lack of "enough" light. It also seems to be the reason Jade not to recognize QR codes created only with dots.

JamieDriver, 2023-05-18:
Not sure what you mean - do you mean adjust the brightness of jade's screen/display (this would affect QRs Jade is showing - it wouldn't affect Jade scanning QRs in a dark environment), or do you mean look at increasing the light-sensitivity of the camera ?

The former is being added in `0.1.48`, but I think you're probably talking about the latter ?

dangarthwaite, 2023-12-06:
This feature is present and vital on the seedsigner when using the Waveshare 1.3" LCD. 

It is super bright by default and drowns out the receiving camera.  I've spent two hours over as many days trying to get sparrow on my laptop to read the QR code on my DIY esp32 + waveshare 1.3" LCD, no success.

JamieDriver, 2023-12-07:
Hi,
For official Blockstream Jade devices (other than the first small run) there is a menu option to control screen brightness - this is feasible because of the way we specifically had the hardware made/wired.  eg. the axp power chip has an output powering the screen, and we can control the voltage of that output.
With generic esp32 devices we don't know exactly how it's wired/configured - indeed most generic devices don't have an axp to control the power, and the screen will just have a fixed voltage - there's nothing we can do in software in that case.

Saying that, we can mock 'dimming the screen' for QR codes by using some darker grey colour rather than full-on white for the 'white' parts of the QR - in fact we do this already and use a pale grey rather than full white ... do you think this would be sufficient ?
ie. if there was a control for how pale/dark a grey you want to QR 'white' areas to be ?