Blockstream/Jade #57: Please add monochrome display option

danskich, 2023-03-18:
There's no option to switch between monochrome or color display. I noticed after several wallets switched from monochrome to color display, it got way brighter and 'tough' on the eyes, especially if you need to do multiple transactions or use the wallet for extended periods of time. Monochrome would be helpful option to have.

JamieDriver, 2023-05-18:
Not monochrome, but adding a 'screen brightness' setting in fw `0.1.48`, so you can dim the screen to your preferred brightness (side effect - dimming screen should increase battery life).

JamieDriver, 2023-07-17:
FYI: intend to offer this in the upcoming fw release.  There will be an option to change the highlight colour used - as well as offering other 'bright' colours, there will also be light/dark grey options - selecting one of these should effectively be a 'monochrome mode'.

closed 2023-08-02 by JamieDriver