Blockstream/Jade #59: Issue using personal blind oracle with webpage (emscripten esp32 bc-ur)

Ciugam, 2023-03-30:
Hi,
I have a problem during the first setup of my personal webpage with Jade.
Here are the steps I've done to explain at my best:
1) I set up my personal blind oracle and the esp32 webpage using emscripten.sh script;
2) I've exposed both oracle and webpage with 2 different onion services, to reach them outside;
3) I've done a factory reset of Jade and I've pointed it to my personal blind oracle's onion service (+port);
4) All fine; Jade asked: "how do you want to connect?" I chose QR;
5) I set up first my pin, repeat to confirm;
6) I went to my onion webpage (qrpin.html) and I scanned the qr from my Jade;
7) The webpage remained stuck ad "step 1 of 4" and did not reply with its qr code.

If I skip the 3rd step (not point to blind oracle, and using default server), all goes fine and I can use my webpage to unlock the Jade.
If I don't use the webpage, I can unlock Jade with my blind oracle with cable+companion and all works fine.

How do I fix that? Did I missed something?
Thanks in advance

JamieDriver, 2023-03-31:
> The webpage remained stuck ad "step 1 of 4" and did not reply with its qr code.

You mean Jade was stuck on 'scanning' ?   ie Jade is not liking your personal server QR ?  It is still scanning and not displaying a QR for the webpage to scan ?  And with the default server QRs it's fine ?

3rdIteration, 2023-03-31:
It's probably a HTTPS issue. Did you include a http or HTTPS prefix with your pinserver?

If you open the developer console for your browser, it is likely trying to access your server over http. 

You should also try to unlock your Jade using Electrum or Sparrow, as they might have issues too depending on whether you included a http or HTTPS prefix for your pinserver.

Ciugam, 2023-03-31:
@JamieDriver Jade is not stuck, it's the webpage; it seems that when webpage is scanning Jade's QR, **the webpage itself doesn't like that QR after scan it and so it stucks**. 
So webpage can't proceed with 2nd step, showing the QR that I should scan with Jade.
With default server, and setting up my webpage, all fine. 
The problem starts when I set up Jade to point to my blind oracle **and** also I use my webpage.

To answer @3rdIteration 's suggestion, I've installed a pip3 module called https.server, that simply creates an http webpage with python self-certification TLS. So I don't have any problem with the QR scanning (in fact with Jade's default server and my personal webpage, all goes fine).
Also, using the Jade only with my blind oracle, it works on electrum, sparrow etc. with cable.
Unfortunately, I prefer not to use cables and I'd rather use the airgap functionality with QR unlock.
If you want more infos about "my https version", I'll link you what I've found:
https://github.com/pathtofile/https.server

JamieDriver, 2023-03-31:
Right, so it has scanned the first QR from Jade - which should be Jade telling the web page what its configured pinserver urls are - and then the web page is trying to contact that service.  
Have you tried with a non-onion url for the pinserver - just in case it's a Tor/Onion related issue ?

Can you see if the webpage is trying to make a call to the pinserver (is the url correct), and similarly can you see if the pinserver has anything in its connection logs (ie. is it being reached) ?

Ciugam, 2023-04-01:
I have some news: I set up a dynamic dns and I've port forwarded (no https only http), to test a clearnet address and pointing Jade to the blind oracle.
At first, I've tried to unlock with a cable and Green desktop companion, all fine and super fast.
Next, I've tried to run the webpage, result: I've got the same situation as the first message: webpage stucks at "step 1 of 4".
In the console debug page, "failed to hit pin server url".

JamieDriver, 2023-04-03:
```code 501, message Unsupported method ('POST')```

Looks odd - all calls to the pinserver are expected to be POSTs, iirc.  port number (8096) def correct ?  No proxy/gateway etc in between which might be getting this and rejecting the POST ?

JamieDriver, 2023-04-03:
Does your pinserver look like the same domain as the webserver ?  Just thinking about cross-site scripting issues ... (??)
Sorry, clutching at straws ...

Ciugam, 2023-04-03:
@3rdIteration thanks! I've tried to set up jade using url http://###.onion:8096, and then using regular "python3 -m http.server 9999" it works!
Unfortunately, When trying to build the https website with self-signed cert, it doesn't work.
Also, because of http, onion site of the webpage doesn't grant me camera permissions, and also in http LAN it has the same problem. I could say that it worked because of the chromium flag "insecure origins treated as secure" in which I put local IP of the machine.
As @3rdIteration said, the problem could be how the webpage is hosted. If it is in http, it can redirect to onion service pinserver, but http is not enough to expose on the web, also with onion service I cannot get camera permissions.
If the webpage is "launched" with https, then it has trouble to redirect to the http pinserver (onion or clearnet).

To @JamieDriver: I've checked some options, but my IP has not NAT issue (actually I host and expose other things through port-forwarding, so no problem).
The usb has never been have any issues, it works every url I point to Jade, clearnet or onion. Tested on Sparrow, Electrum, Green (Desktop) and Specter Desktop.
Also, no cross-site issues; I set up 2 completely different onion services to pinserver and webpage. Also tried with clearnet, I've exposed only pinserver with http and port-forwarding: no issues on the pinserver itself.

3rdIteration, 2023-04-03:
You can get away with http if you self host the PinQR tool locally, especially if you set the Jade pinserver address to localhost and just use something like SSH to forward the ports. (Meaning that you don't need to expose your remote pinserver directly to the internet, nor will your Jade leak any meaningful information about the location of the pi server) At this time that requires manually downloading it the pinQR helper webpage serving the content locally. (I have an open issue on the pinserver repo for this reason)

Ciugam, 2023-04-04:
@3rdIteration many many thanks! 
I used tor to local port forward the 8096 port from my remote machine to laptop.
Then I downloaded the repo on my localhost laptop, run emscripten.sh and I've launched the webserver using python in http.
So, if I follow this steps it works perfectly!
Probably the best thing should be to put all this toghether, like you said in the blind pinserver repo issue: merge the blind oracle and the webpage helper, and try to communicate via tor and https connection.

Ciugam, 2023-04-04:
@JamieDriver I didn't push to jade any cert.
In fact the blind oracle connects only via http: no certs, no https. Blind oracle runs in my tests in http clearnet or onion http address.
I want to highlight that in this situations, using the cable it works flawlessy, so blind oracle is not the problem.

The issue is the communication between webpage and blind oracle.
If I use webpage and blind oracle on remote machine, and using http connections, browsers can't grant me camera permission so I can't unlock the Jade (security reasons - http is not secure on remote).
**If I https only the website** (not the blind oracle), I can access to camera permission but webpage can't communicate to blind oracle (don't know why).

As @3rdIteration said, to me a temporarily solution is to local port forward the remote blind oracle to my laptop, and run webpage on laptop itself. So all works in http and webpage is in my localhost, so no troubles with grant camera permissions.