Blockstream/Jade #62: Left button for PIN entry not functioning on Windows USB for TTGO Jade wallet DIY

bayfitt, 2023-04-27:
Description:
I would like to report a bug for the TTGO Jade wallet DIY. I have tested the wallet on USB on Windows, Linux, and Android. When I went to enter my PIN, the left button on the TTGO did not work on Windows, while the right button was functional. I have a video recording of the issue, which I can share with the developers. I consistently experienced this issue even after restarting both the wallet and the computer on Windows. On the other hand, on Linux and Android on USB, everything worked fine. Additionally, Bluetooth on Android worked as expected, and both buttons functioned correctly when entering my PIN. 

Details:
- Issue: The left button on the TTGO did not work when entering my PIN on Windows
- Reproduction Steps: 
    1. Power on the TTGO Jade wallet DIY
    2. Connect it to a Windows computer via USB
    3. Try to enter your PIN using the left button
- Expected Result: The left button should work correctly, allowing me to enter my PIN.
- Actual Result: The left button does not work on Windows, preventing me from entering my PIN.
- Video Evidence: I have a video recording of the issue, which I can share with the developers.
- Firmware version: 0.1.47-19-ga509904
- Secure Boot enabled: Yes

Additional Notes:
- The issue was not present on Linux and Android on USB.
- Bluetooth on Android worked as expected, and both buttons functioned correctly when entering my PIN. 
- I would appreciate it if the developers could investigate this issue and provide a fix as soon as possible.

bayfitt, 2023-04-28:
Uploading 1_5064448508560409232.mp4 …

Uploading 1_5064448508560409231.mp4 …

bayfitt, 2023-04-28:
https://user-images.githubusercontent.com/26312519/235023414-4ffc3ad7-0570-41b2-a3c6-fb125aa2d82f.mp4

bayfitt, 2023-04-28:
https://user-images.githubusercontent.com/26312519/235023914-9346cd3d-23ee-4c4e-b96f-409a112bdc43.mp4

https://user-images.githubusercontent.com/26312519/235023943-e093a299-16df-4911-be84-f6a65dcbde7a.mp4

jpph, 2023-04-29:
As a temporary workaround, you can use electrum for windows to unlock the ttgo display, then close electrum, then launch Blockstream Green for windows.

bayfitt, 2023-04-30:
My concern was not about it not working in windows. I can use Linux or Android. My concern is that it works differently, which means there is something going inside the computer from that wallet. Why should it work differently in different OSes, if it is supposed to be isolated? To me, this is a security concern.

jpph, 2023-04-30:
As per cryptoguide told me there is some simplified input mode when connected to other software than  green wallet. Seems that android and Linux version are using the simplified mode

JamieDriver, 2023-05-02:
This sounds v odd to me - you are not re-flashing the device between using on windows and linux, right ?
- You plug it into linux, and can use both buttons and unlock the device with PIN
- You plug the same device/same fw etc. into windows, and the left button does not work ?

This is certainly not expected.  
If I had to hazard a guess I'd say that maybe there is some pin/interrupt conflict that windows is triggering which is somehow disrupting detecting the button press.  ('pin' as in physical pin on the board, not as in 'personal id number'!)

JamieDriver, 2023-05-02:
So you plug a lilygo t-display into Green Desktop on Windows, and the left button does not work ?
(iirc the right and left buttons should scroll the numbers up/down, and you press both together to move to 'next digit' ?)

We will certainly investigate - as you say the unit buttons should not be affected by whatever happens to be on the other end of the USB cable!

jpph, 2023-05-02:
I have same behavior that on the video. At first digit left button act as ''both button'', then at second digit it stop working. And both button stop working too. Very strange behavior

JamieDriver, 2023-05-02:
Looks like something specific to Green Desktop (on windows) rather than an issue with the Jade ttgo fw per se.  (eg. setting some usb/serial flags badly, voltage on wrong pins, etc.).
We'll investigate further - thanks for bringing it to our attention!

sandman21vs, 2023-05-29:
I'm having exactly the same problem, the tip to connect the electrum was good, I hadn't thought of that, thanks.
I hope they fix this bug soon and it's not a security breach

JamieDriver, 2023-05-29:
Are you running the fw with BLE/radio enabled ?

sandman21vs, 2023-06-12:
I'm running directly connected to the cable

JamieDriver, 2023-06-12:
yy, but I was wondering if Bluetooth was running/enabled, even if youre not connected by Bluetooth - just in case it was relevant ...  (if 'Yes' - does it make any difference if you disable it ?)  
This is all info I can pass to the desktop guys.

jpph, 2023-06-12:
For myself, I have the issue with cable connected and Bluetooth disabled.

sandman21vs, 2023-06-13:
Both with Bluetooth on and off, jade diy does not work in the blockstream app (but only in the Green app, in other apps it works perfectly)

JamieDriver, 2023-06-13:
Cheers.  I'll pass the details along to the windows/desktop app guys.  
We'll also check the build/button config to see if we've got usb/serial signals and button signals getting tangled.

Are you all on lilygo-tdisplay units ?

sandman21vs, 2023-06-20:
yes we are using liligo tdisplay, in a few days my camera arrives and then I'm going to make a diy jade with it too

adri76, 2024-07-30:
I can confirm that I have the same issue. It is only a problem when using Green wallet via USB. Bluetooth works fine.

Sandmann21s, 2024-07-30:
> Posso confirmar que tenho o mesmo problema. É apenas um problema ao usar a carteira verde via USB. Bluetooth funciona bem.

I can't use Jade via BT on Windows, but on Linux the blockstream green app works well via cable 

Another problem is that the battery indicator does not work showing the device charge

JamieDriver, 2024-07-31:
> Another problem is that the battery indicator does not work showing the device charge

Yes, that hw board has no chip telling us that sort of information.

Sandmann21s, 2024-08-28:
[image]
This issue has been resolved, and now the button works correctly on the Jade DIY. Thank you for the fix!

Sandmann21s, 2024-08-30:
Through trial and error I discovered that if you format your DIY Jade, open the Green app in Windows, connect the Jade, and do the setup it starts to work, but there is no way to connect a Jade that you did not setup in Windows, (the bug does not persist if you connect the Jade after the Green app opens, but if the Jade is not configured Windows does not try to connect to it, when it tries to connect it gives an error lol)

Just to comment, I tested it with an M5stack C2Plus and it works normally, apparently it is only the TTGO T Display that has this problem, the sad part is that this is the most used DIY device to make DIY Jade

adri76, 2024-08-30:
I tried again, but could not get it to work with green wallet.
It works with other wallet apps, but not green wallet. (no way to enter the pin)
The point of a hardware wallet is to securely store your keys, and I would not trust a buggy setup to store anything of value, so it is nice to use as a proof of concept, but it defeats the object of a diy wallet.

Sandmann21s, 2024-08-30:
What's interesting is that this issue only occurs in the Windows version. I believe it would be helpful to open an issue on Green's GitHub to get it resolved because it works fine on Android, macOS, and Linux—only the Windows version is affected.

sandman21vs, 2024-12-06:
The previously reported bug has changed at this time, I don't know exactly what is happening but Jade now clicks both blocks alone once when the green wallet is opened and after that it keeps clicking the 

https://github.com/user-attachments/assets/b71c0922-a600-4f1c-ba2b-7a5270377227

top button infinitely and does what appears in the video above