Blockstream/Jade #63: Unclear error message - Unhandled QR Payload

bitcoinprecept, 2023-05-01:
Hello Team - I want to highlight an opportunity to IMHO improve the UI around SeedQR flow, specifically around one button and one error message.

I am testing the stateless, airgapped, SeedQR flows in firmware 0.1.45.

After scanning in a SeedQR and resting on the main wallet fingerprint screen, the "QR Scan" menu choice was not intuitive to me. Is this where I scan a PSBT? Can I add multiple seeds as is the case in Seedsigner? Does this button only allow me to switch to a new seed and PSBT flow is somewhere else? What other reasons might I scan a QR code right now?

Not being sure, I thought I'd test scanning in a different SeedQR at this point. I received an error code that is surely technically accurate, but I feel there's an opportunity to use the screen real estate here to guide the user into the workflows the Jade is expecting:

[image]

I don't know what all the possible or supported workflows are here, but I'm generally most familiar with PSBT signing. If that's the intended flow here, perhaps an error message of "Error - No PSBT identified in QR code" would be clearer to the user. Of course if that's the only possible flow, I might argue you'd also help the user by re-naming the menu option from "QR Scan" to "Scan PSBT".

Logic to identify that a new seed has been scanned might also be useful here. Potential feedback message "New seedphrase detected - would you like to switch to new wallet?"

Overall great work to see this SeedQR functionality running on the Jade! Thank you!!!

bitcoinhelp, 2023-05-01:
Thanks for the feedback! Here's a list of all supported QRs for the "QR Scan" function: https://help.blockstream.com/hc/en-us/articles/14817768566297-Which-types-of-QR-codes-are-supported-by-QR-Scan-

1.  PSBTs
2. Addresses for validation
3. Multisig configurations
4. Message signing with specter desktop

Jade doesn't currently allow for 2 wallets to be available at the same time, however the "New seedphrase detected" idea is interesting if technically feasible - where Jade uses the 'logout' function behind 'session' and then opens your new wallet, or at the very least lets you know you need to logout first and rescan your SeedQR

JamieDriver, 2023-05-18:
Yes, we could look to add recognising the supported 'seed' formats here and doing something as suggested above.

JamieDriver, 2023-07-17:
FYI intend to handle this case in next fw release.  ie. if you scan a wallet/seed qr, it will pop up a dialog asking you to confirm that you want to switch wallets.

Marc-Gee, 2023-07-18:
Great work by you all on the QR codes!!

JamieDriver, 2023-08-02:
Jade fw `1.0.21` should accept a wallet/seed QR from the generic 'Scan QR' screen, and ask if you want to switch wallets.  
Closing as addressed - please reopen if this is not what was intended / is not sufficient.  
Thanks.

closed 2023-08-02 by JamieDriver