Blockstream/Jade #65: docker-compose build fails for TTGO T-DISPLAY

giovantenne, 2023-05-18:
```
(local)$ git clone --recursive https://github.com/Blockstream/Jade.git
(local)$ cd Jade
(local)$ docker-compose up -d
(local)$ docker-compose exec dev bash
(docker)$ cp configs/sdkconfig_display_ttgo_tdisplay.defaults sdkconfig.default
(docker)$ git submodule update --init --recursive
(docker)$ idf.py build
```

Here is the logs:

```
[1123/1193] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/sensitive.c.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/sensitive.c.obj 
/root/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DBUILD_ELEMENTS=1 -DBUILD_MINIMAL=1 -DHAVE_CONFIG_H -DHAVE_MBEDTLS_SHA256_H -DHAVE_MBEDTLS_SHA512_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../main -I../main/ble -I/root/esp/esp-idf/components/newlib/platform_include -I/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/root/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/root/esp/esp-idf/components/freertos/esp_additions/include -I/root/esp/esp-idf/components/esp_hw_support/include -I/root/esp/esp-idf/components/esp_hw_support/include/soc -I/root/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/root/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/root/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/root/esp/esp-idf/components/heap/include -I/root/esp/esp-idf/components/log/include -I/root/esp/esp-idf/components/soc/include -I/root/esp/esp-idf/components/soc/esp32/. -I/root/esp/esp-idf/components/soc/esp32/include -I/root/esp/esp-idf/components/hal/esp32/include -I/root/esp/esp-idf/components/hal/include -I/root/esp/esp-idf/components/hal/platform_port/include -I/root/esp/esp-idf/components/esp_rom/include -I/root/esp/esp-idf/components/esp_rom/include/esp32 -I/root/esp/esp-idf/components/esp_rom/esp32 -I/root/esp/esp-idf/components/esp_common/include -I/root/esp/esp-idf/components/esp_system/include -I/root/esp/esp-idf/components/esp_system/port/soc -I/root/esp/esp-idf/components/esp_system/port/include/private -I/root/esp/esp-idf/components/xtensa/include -I/root/esp/esp-idf/components/xtensa/esp32/include -I/root/esp/esp-idf/components/lwip/include -I/root/esp/esp-idf/components/lwip/include/apps -I/root/esp/esp-idf/components/lwip/include/apps/sntp -I/root/esp/esp-idf/components/lwip/lwip/src/include -I/root/esp/esp-idf/components/lwip/port/esp32/include -I/root/esp/esp-idf/components/lwip/port/esp32/include/arch -I../components/assets -Iesp-idf/assets -I../components/libwally-core/upstream/include -I../components/tft -I../components/spidriver -I/root/esp/esp-idf/components/driver/include -I/root/esp/esp-idf/components/driver/deprecated -I/root/esp/esp-idf/components/driver/esp32/include -I/root/esp/esp-idf/components/esp_pm/include -I/root/esp/esp-idf/components/esp_ringbuf/include -I../managed_components/espressif__libsodium/libsodium/src/libsodium/include -I../managed_components/espressif__libsodium/port_include -I/root/esp/esp-idf/components/mbedtls/port/include -I/root/esp/esp-idf/components/mbedtls/mbedtls/include -I/root/esp/esp-idf/components/mbedtls/mbedtls/library -I/root/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I../components/button -I../components/button/include -I../components/esp32-rotary-encoder/include -I../components/esp32-quirc/lib -I../components/esp32-quirc -I/root/esp/esp-idf/components/bootloader_support/include -I/root/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/root/esp/esp-idf/components/app_update/include -I/root/esp/esp-idf/components/esp_app_format/include -I/root/esp/esp-idf/components/esp_partition/include -I/root/esp/esp-idf/components/nvs_flash/include -I/root/esp/esp-idf/components/esp_timer/include -I/root/esp/esp-idf/components/esp_wifi/include -I/root/esp/esp-idf/components/esp_event/include -I/root/esp/esp-idf/components/esp_phy/include -I/root/esp/esp-idf/components/esp_phy/esp32/include -I/root/esp/esp-idf/components/esp_netif/include -I../components/autogenlang -Iesp-idf/autogenlang -I../managed_components/espressif__cbor/port/include -I../components/esp32_bsdiff -I../components/esp32_deflate -I../managed_components/espressif__nghttp/port/include -I../managed_components/espressif__nghttp/nghttp2/lib/includes -I../components/esp32_bc-ur/src -I/root/esp/esp-idf/components/http_parser -I/root/esp/esp-idf/components/efuse/include -I/root/esp/esp-idf/components/efuse/esp32/include -I/root/esp/esp-idf/components/esp_eth/include -I../managed_components/espressif__esp32-camera/driver/include -I../managed_components/espressif__esp32-camera/conversions/include -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/jade=. -fmacro-prefix-map=/root/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0.1\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/sensitive.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/sensitive.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/sensitive.c.obj -c ../main/sensitive.c
../main/sensitive.c:16:2: error: #error "Error, CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS should be increased"
   16 | #error "Error, CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS should be increased"
      |  ^~~~~
[1131/1193] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/selfcheck.c.obj../main/selfcheck.c:751:13: warning: 'test_bcur_large_payload_many_icons' defined but not used [-Wunused-function]
  751 | static bool test_bcur_large_payload_many_icons(void)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[1132/1193] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/wallet.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /jade/build/log/idf_py_stderr_output_970 and /jade/build/log/idf_py_stdout_output_970
root@5441bc3fc840:/jade#
```

JamieDriver, 2023-05-22:
Agree - reproduced.
```
root@dea780e4dbd0:/jade# cp configs/sdkconfig_display_ttgo_tdisplay.defaults sdkconfig.default
root@dea780e4dbd0:/jade# rm sdkconfig
root@dea780e4dbd0:/jade# rm -fr build
root@dea780e4dbd0:/jade# idf.py reconfigure
root@dea780e4dbd0:/jade# grep CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS sdkconfig.default 
CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=3
root@dea780e4dbd0:/jade# grep CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS sdkconfig         
CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=1
```

Aha!  
`cp configs/sdkconfig_display_ttgo_tdisplay.defaults sdkconfig.default` is incorrect - that should be 'sdkconfig.default**s**'
Then all is good ;-)

If there is an error in our documentation anywhere, please do let us know.
Otherwise, if you agree all is good with that typo correction, please close this item.  
Many thanks.

giovantenne, 2023-05-22:
My bad :-)
All good

Many thanks

closed 2023-05-22 by giovantenne

JamieDriver, 2023-05-22:
(just fyi I'll fix some of those compilation warnings about functions defined but not used - nicer to keep it clean!)