Blockstream/Jade #67: xpub import - invalid base58 character 0x3a

scarolan, 2023-05-28:
I get this error when attempting to use Xpub export on the Jade device, and import it using Blockstream Green.

Steps to reproduce:

* Create standard wallet on Blockstream Jade
* Go into Options > Wallet  > Xpub Export
* Make sure singlesig and native segwit are selected
* On the Android Green app select "Watch-only" during setup, select singlesig
* Select Singlesig, scan QR code animation with phone camera
* Click "Log In", get error message:

"Invalid base58 character 0x3a"

I'm trying to set this up using the instructions here:
https://help.blockstream.com/hc/en-us/articles/11243000549401

JamieDriver, 2023-05-29:
Hi,
When we added QR xpub export to Jade it was decided to use a format that was supported by the majority of 3rd party wallet apps, as the main goal here was to improve compatibility with existing QR-friendly wallets (eg Sparrow, Specter, Nunchuk, BlueWallet, etc).  (format: `ur:crypto-account`, from Blockchain Commons).   
Unfortunately this format is not yet supported by the Green apps - although support is planned and is I believe in progress.  (The next GDK release should contain the underlying libraries needed to create and parse the BC UR formats.)  

I will update this item as this work progresses, to keep you informed.

scarolan, 2023-05-29:
Thanks for the update, @JamieDriver .  I'll try it out with Nunchuk wallet today.

The Jade wallet is awesome, love the camera feature for air gap functionality.

JamieDriver, 2023-06-02:
Hi @samsam380,  
Always happy to help!  
So an not to obscure this item about the QR xpub export, I've replied on new issue #69 dedicated to this discussion.

closed 2024-01-28 by scarolan