Blockstream/Jade #69: Discuss Jade Entropy source

JamieDriver, 2023-06-02:
Replying to: https://github.com/Blockstream/Jade/issues/67#issuecomment-1572153756

JamieDriver, 2023-06-02:
Always happy to help!  

Short answer:  
See [this web post](https://help.blockstream.com/hc/en-us/articles/9640569620761-How-does-Blockstream-Jade-generate-it-s-recovery-phrase-)

Long answer:  
If you opt for new wallet on Jade (ie for Jade to create you a new wallet on-board the hw - always the best option!), [this function](https://github.com/Blockstream/Jade/blob/eeaeb1103ba0e02ef05f778ba13f7846474e6520/main/keychain.c#L256) is called.  
As you can see it gets some entropy whose length depends on how many words you want in your recovery phrase (12-words/128bits is usually considered sufficient, especially if paired with a passphrase, but we also offer 24-words/256bits).

The entropy is obtained by calling [`get_random()`](https://github.com/Blockstream/Jade/blob/eeaeb1103ba0e02ef05f778ba13f7846474e6520/main/random.c#L136) which gets entropy from the internal random machine.  

When Jade starts up, the random machine is seeded with initial entropy from a number of sources/sensors on the chip - temperature, battery/usb current/voltages, radio noise etc. - these environment sensors themselves include a degree of random 'noise' as well as whatever they actually sample.   See [here](https://github.com/Blockstream/Jade/blob/eeaeb1103ba0e02ef05f778ba13f7846474e6520/main/random.c#L65).  (Ignore the incorrect function comment!  I will fix asap!!)  

Also during boot the camera is activated and a number of frames of image data is fed into the random seeding - again even is jade was lying on a desk, the 'black frames' captured would contain random noise.  See the boot up code [here](https://github.com/Blockstream/Jade/blob/eeaeb1103ba0e02ef05f778ba13f7846474e6520/main/main.c#L59) and [here](https://github.com/Blockstream/Jade/blob/eeaeb1103ba0e02ef05f778ba13f7846474e6520/main/main.c#L136).

During operation, the internal random machine frequently re-fetches source entropy from the sensors on the chip - temperature, battery/usb current/voltages, radio noise etc. as above (but not the camera) to feed into the machine.  See wherever [refeed_entropy()](https://github.com/Blockstream/Jade/blob/eeaeb1103ba0e02ef05f778ba13f7846474e6520/main/random.c#L129) is called.

Additionally at any point the connected application can send in some entropy which jade also throws into its mix of seed entropy.  [See here](https://github.com/Blockstream/Jade/blob/eeaeb1103ba0e02ef05f778ba13f7846474e6520/main/process/dashboard.c#L276)  (Apps would usually feed in some entropy when they initially connect.)

NOTE: when source entropy is feed into the machine it is always cumulative/additional.  We are always 'adding' some more seed randomness, not 'replacing' old data with new data.

I hope this helps.

samsam380, 2023-06-03:
@JamieDriver Thank you so much!

michaelfolkson, 2023-06-03:
@JamieDriver: Nice post. Added to [Bitcoin StackExchange](https://bitcoin.stackexchange.com/questions/118535/how-is-the-entropy-created-for-generating-the-mnemonic-on-the-jade-hardware-wall).

JamieDriver, 2023-06-26:
Moved to [wiki](https://github.com/Blockstream/Jade/wiki/Jade-Entropy)

closed 2023-06-26 by JamieDriver