Blockstream/Jade #71: Difference between sleep and logout?

scarolan, 2023-06-11:
I couldn't find any info in the FAQ or docs.  What is the difference between putting the device into deep sleep vs using logout?

As far as I can tell the behavior is identical.  To wake the device or power it back on I still have to push the green button and re-enter my PIN.

bitcoinhelp, 2023-06-12:
Logout won't shut off your device, just bring you back to the "home"/connect Jade screen

scarolan, 2023-06-12:
So how do you wake it from sleep then?  I tried short presses of each button, the only thing that worked to get the screen back was long-press the green button.

scarolan, 2023-06-12:
Why is it called 'sleep' and not 'power off'?  Is there even a difference?

scarolan, 2023-06-12:
Roger that, thanks!  Closing issue.

closed 2023-06-12 by scarolan