Blockstream/Jade #73: electrum 4.4.0-4.4.2 not working with jade-diy bitcoin app (pre-2.1): no sig for <compressed_pubkey>

sandman21vs, 2023-06-13:
the same problem is happening on the link that I will send below, in the electrum wallet jade cannot make transactions, according to the electrum staff it is because it is probably running the old version of something related to bitcoin ,
I hope this bug is fixed (and also the bug that prevents using jade diy on windows with the green wallet app)
https://github.com/spesmilo/electrum/issues/8365

sandman21vs, 2023-06-13:
[image]

JamieDriver, 2023-06-19:
This is a regression issue with electrum wallet, and was resolved by commit: https://github.com/spesmilo/electrum/commit/2752cd21834f81af4b71b80277191271f98f6518

See: https://github.com/spesmilo/electrum/issues/8463

My understanding is that it is fixed in: https://github.com/spesmilo/electrum/releases/tag/4.4.5

sandman21vs, 2023-06-20:
Thanks, I'm going to compile and test, it looks like this release came out 7 hours ago and it's still not available on the electrum website to download

JamieDriver, 2023-06-20:
Super - pls close this issue if their fix resolves the problems you are experiencing.
Many Thanks.

JamieDriver, 2023-06-30:
Is this resolved ?  Can we close ?  Thanks ?

JamieDriver, 2023-08-02:
Closing due to inactivity.  Believed to be caused by a now-resolved Electrum issue.

closed 2023-08-02 by JamieDriver