Blockstream/Jade #76: Getting an error while trying to flash TTGO

bitcoinfixesalot, 2023-07-24:
[image]

Am I doing something wrong?

JamieDriver, 2023-08-02:
As far as I can see you're not doing anything wrong.  
Is this the latest jade master branch (with the `configs/sdkconfig_display_ttgo_tdisplay.defaults` copied over `sdkconfigs.defaults`) ??
Have you tried removing `sdkconfig` and `idf.py fullclean all` first ?
I'm clutching at straws, to be honest.

JamieDriver, 2023-08-02:
Hm, ok:
`DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html`

I don't see that ... might mean your python version is different which *may* affect those hashes.

FYI: I'm using Python 3.9.2, and have different environments with setuptools versions `52.0.0`, `67.4.0` and `67.8.0` - all of which seem fine.  ( `python -c "import setuptools; print(setuptools.__version__)"`)  (*I note the latest is `68.0.0` ...)

JamieDriver, 2023-08-02:
Also, probably unrelated to this precise issue but:  
`Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.`

Not ideal - won't know what version tag to apply.  Better to ensure the Jade dir is a git repo and the user building has relevant permissions.

JamieDriver, 2023-09-20:
Is this resolved ?