Blockstream/Jade #79: Question: Firmware verification

rstmsn, 2023-08-02:
When updating a Jade hardware wallet, does the device carry out any sort of signature verification, to ensure the FW build has come from a trusted source, ie Blockstream?

JamieDriver, 2023-08-04:
Hi,
If you have a standard Jade unit (eg. as purchased from Blockstream store) then yes, the unit will verify the firmware uploaded during the upgrade process has been signed by Blockstream.  In fact it carries out this check every time it boots the firmware.  
This security feature is provided by the esp-idf 'operating system' that underlies the Jade firmware - it is usually referred to as 'secure boot' and you can check the technical details [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html).  

Also, during the upgrade process the unit should display the 'hash' of the data file being uploaded - we intend to also publish a web-page listing the good/valid hashes, although this is still a work-in-progress atm.

NOTE: if you have a 'diy' unit (eg. you have purchased a consumer esp-32 board, like a lilygo t-display or an m5stack unit) and have loaded the firmware yourself, the 'secure boot' feature is not enabled, as you are in full control and are able to load any firmware you desire - eg. one you have modified yourself.  (Of course in that case you are free to enable 'secure boot' such that it only accepts firmware signed by you!)

JamieDriver, 2023-10-06:
Closing as inactive.
Please comment/reopen if further clarification is needed.

closed 2023-10-06 by JamieDriver

JamieDriver, 2024-08-20:
We could possibly read the fw partition and compute the hash and show the user.

The problem is that a fake/malicious fw pretending to be version X would just discard the hash it computed and instead display the (hard-coded) correct hash for version X - so I'm not sure it gains you much, security-wise.

Tacitoku, 2024-08-21:
Thank you very much for your analysis. 

You brought up an important point that I hadn't thought of. 
It doesn't make sense to show the hash value on the Jade display after the firmware has been updated, because that would possibly lull the user into a false sense of security.