Blockstream/Jade #80: Running ./update_jade_fw.py on Mac M1 just crashes

mikeziri, 2023-08-04:
Hi,
I can't seem to run the repo's `./update_jade_fw.py` python command.
I'm not very familiar with python.
It crashes with `ModuleNotFoundError`.
It seems like the repo needs some dependencies.
Read on it and I think one should run `pip install -r requirements.txt`.
The `requirements.txt` is available but upon running it crashes with:
`ERROR: No matching distribution found for wallycore==0.8.9`
If it helps, I'm running on a Mac M1 with python3 installed via homebrew.
Might try to run that on a Linux machine later but would love to see some better guide on this self upgrade.
There's really not much info on this "guide": https://github.com/Blockstream/Jade/blob/master/FWUPDATE.md

JamieDriver, 2023-08-07:
```ERROR: No matching distribution found for wallycore==0.8.9```  
It sounds like libwally does not support this platform.  

The wally release notes for `0.8.5` includes the entry : "Add support for M1 macOS arm64 and universal2 builds/Python wheels" and I note that `0.8.9` includes these artifacts: `wallycore-0.8.9-cp310-cp310-macosx_11_0_arm64.whl` and ` wallycore-0.8.9-cp310-cp310-macosx_10_9_universal2.whl `, so looks to me like it should support python10+osx11 (and maybe osx10.9?)

Do you know what osx version you're on, and what version of python you are running (`python --version`) ?

Fair point re: mentioning the requirements file in the guide.

jgriffiths, 2023-08-07:
@JamieDriver @mikeziri 

The following commands will let you know if the latest wally 0.9.1 is compatible with the box:

```
$ virtualenv venv-tmp
$ . venv-tmp/bin/activate
$ pip install wallycore
# To cleanup
$ deactivate
$ rm -rf venv-tmp
```

If that doesn't work please give the python version with `python --version`.

mikeziri, 2023-08-07:
> Do you know what osx version you're on, and what version of python you are running (`python --version`) ?

I'm on MacOS 13.5 (22G74) Ventura (M1)
Python 3.11.4
virtualenv 20.24.2

> ```
> $ virtualenv venv-tmp
> $ . venv-tmp/bin/activate
> $ pip install wallycore
> # To cleanup
> $ deactivate
> $ rm -rf venv-tmp
> ```

Those commands run alone work. it installed wallycore 0.9.1.

Trying `pip install -r requirements.txt` that, again, I'm assuming one is suppose to do as there is no documentation on how to use the `update_jade_fw.py` command, it outputs:
```
ERROR: Could not find a version that satisfies the requirement wallycore==0.8.9 (from versions: 0.8.1, 0.8.2, 0.9.1)
ERROR: No matching distribution found for wallycore==0.8.9
```

jgriffiths, 2023-08-07:
@mikeziri Agreed on the usability aspect. libwally, and by extension Jade, need to make updates whenever new versions of Python come out. In this case Python 3.11 was just released, and support for it added to wally last week. There will always be a slight lag from Python releases to wally to Jade, since we need to update and then QA/test each update before we can release each dependent component. Unfortunately you happened to grab the latest Python from homebrew in between the time we could update on our side. If its any consolation, this should be rare occurrence.

Jamie has just merged an update here to use the latest wally with Python 3.11 support (https://github.com/Blockstream/Jade/commit/61cf21543b49fb9d7b4b94f0a073d1a015a55875). So, if you sync master you should find everything works as expected now. Thanks for reporting.

JamieDriver, 2023-08-08:
As Jon says - we have just added support for python 3.11 and it will be included as standard in the next Jade release - in the meantime if you sync your jade repo to the latest tip of the 'master' branch that should work (as includes those changes).
```
git checkout master
git pull
```

Yes, I will review the FWUPDATE.md file and ensure the dependencies install is documented - thanks for that.  (I will leave this issue open until I have done that.)

FYI, if you are using a browser that supports 'WebSerial' you can also use this site for updating the firmware:
https://jadefw.blockstream.com/upgrade/fwupgrade.html

JamieDriver, 2023-09-20:
Should be addressed in fw `1.0.23`

islandbitcoin, 2023-10-10:
I'm having the same problem on my M1. 
I tried to use my linux to update instead. doing the `pip install -r requirements.txt` installed all the dependencies successfully, but then when running `./update_jade_fw.py` I get this error:

```
MacMax:Jade dread$ ./update_jade_fw.py
Traceback (most recent call last):
  File "/Users/dread/Documents/Island-Bitcoin/Finances/Jade/./update_jade_fw.py", line 244, in <module>
    fwlen, patchlen, fwhash, fwcmp = download_file(verinfo, release)
  File "/Users/dread/Documents/Island-Bitcoin/Finances/Jade/./update_jade_fw.py", line 76, in download_file
    hw_target = {'JADE': 'jade', 'JADE_V1.1': 'jade1.1'}.get(verinfo['BOARD_TYPE'])
KeyError: 'BOARD_TYPE'
```

Here is the debug logs:
```
args: Namespace(serialport=None, release=None, fwfile=None, loglevel='DEBUG')
Sending: {'method': 'get_version_info', 'id': '1****3'} as cbor of size 35
Sending: 35 bytes
Sent: 35 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 2 bytes...
Received: 2 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 12 bytes...
Received: 12 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 18 bytes...
Received: 18 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 2 bytes...
Received: 2 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 11 bytes...
Received: 11 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 3 bytes...
Received: 3 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 13 bytes...
Received: 13 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 2 bytes...
Received: 2 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 11 bytes...
Received: 11 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 4 bytes...
Received: 4 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 13 bytes...
Received: 13 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 8 bytes...
Received: 8 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 8 bytes...
Received: 8 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 12 bytes...
Received: 12 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 14 bytes...
Received: 14 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 4 bytes...
Received: 4 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 14 bytes...
Received: 14 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 4 bytes...
Received: 4 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 17 bytes...
Received: 17 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 4 bytes...
Received: 4 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 16 bytes...
Received: 16 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 4 bytes...
Received: 4 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 19 bytes...
Received: 19 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 4 bytes...
Received: 4 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 12 bytes...
Received: 12 bytes
Reading 1 bytes...
Received: 1 bytes
Received msg: {'id': '1****3', 'result': {'JADE_VERSION': '0.1.21', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'JADE_FEATURES': 'SB', 'IDF_VERSION': 'v4.2', 'CHIP_FEATURES': '32000000', 'EFUSEMAC': '2**********8', 'JADE_FREE_HEAP': 2626228, 'JADE_FREE_DRAM': 119468, 'JADE_LARGEST_DRAM': 113804, 'JADE_FREE_SPIRAM': 2506760, 'JADE_LARGEST_SPIRAM': 2506760, 'JADE_HAS_PIN': False}}
Traceback (most recent call last):
  File "~/Jade/./update_jade_fw.py", line 244, in <module>
    fwlen, patchlen, fwhash, fwcmp = download_file(verinfo, release)
  File "~/Jade/./update_jade_fw.py", line 76, in download_file
    hw_target = {'JADE': 'jade', 'JADE_V1.1': 'jade1.1'}.get(verinfo['BOARD_TYPE'])
KeyError: 'BOARD_TYPE'
```

islandbitcoin, 2023-10-10:
That being said, I'm only trying this because my attemps to update the FW using green wallet (tried iOS and Android) and the chrome browser all failed.

JamieDriver, 2023-10-10:
What sort of device are you trying to update there - is this a production jade unit ?

> ./update_jade_fw.py
This is a command for updating a production Jade unit - is this what you have ?  (You do not want to use this to try to update a 'diy' unit you have created yourself from consumer esp32 boards.)

```
{'JADE_VERSION': '0.1.21', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'JADE_FEATURES': 'SB', 'IDF_VERSION': 'v4.2', 'CHIP_FEATURES': '32000000', 'EFUSEMAC': '2**********8', 'JADE_FREE_HEAP': 2626228, 'JADE_FREE_DRAM': 119468, 'JADE_LARGEST_DRAM': 113804, 'JADE_FREE_SPIRAM': 2506760, 'JADE_LARGEST_SPIRAM': 2506760, 'JADE_HAS_PIN': False}
```
Ok - this looks like a very old firmware indeed!  `0.1.21` !   This output does suggest a production unit - one from the first batch run with the proper rotating 'true' wheel.  You are a legend, sir!  ;-)

I suspect that the wallet apps (and indeed that `./update_jade_fw.py` script) no longer handle the 'BOARD_TYPE' being missing (it wasn't output by those old firmwares because there was only one board type in existence at the time).

Many thanks for this report - I will fix `./update_jade_fw.py`  within the next 24-hours, and will raise issues with the application and web teams to get this resolved in their upcoming releases.

Apologies and thanks.

islandbitcoin, 2023-10-10:
Glad we caught the bug!

long time blockstream supporter here... rock on 🫡

JamieDriver, 2023-10-10:
I have been unable to find or immediately create a unit with that software config - I'll have more time tomorrow (already evening in my tz) .  
However, I have fixed the most glaring assumption in the script - if you could `git fetch` the repo and checkout branch `fix_update_fw_script` and try the script from there, you might be in luck (and I'd be much obliged) !

If it's still not working I'll build an old unit tomorrow and investigate further.
Cheers.

JamieDriver, 2023-10-10:
Ok, managed to downgrade to `0.1.21`, fixed script and used it to update to latest.
Good luck!

islandbitcoin, 2023-10-10:
it worked! 🎉

Cheers

JamieDriver, 2023-10-10:
Thanks - apologies for the break in service!  ;-)  I'll get that merged to master tomorrow.  
I hope you enjoy the latest Jade firmware!

JamieDriver, 2023-10-12:
Closing as resolved.  
Please reopen/comment if you are still having issues.

closed 2023-10-12 by JamieDriver