Blockstream/Jade #83: Linker error when flashing TTGO with NORADIO

GebertJung, 2023-08-20:
### Systems: 
- m1 macbook pro/macOS 13.5 (no docker)
- hp zbook 15 g6/debian bookworm xfce (using latest docker-ce as of this writing)

### Steps taken:
exactly as in README except:
- `git checkout 1.0.21 && git submodule update --init --recursive` to set up repo (in Dockerfile.dev, too, otherwise container will miss secp lib)
- `tools/mkdefaults.py config/sdkconfig_display_ttgo_tdisplay.defaults NORADIO` to cp/install sdkconfig.defaults file to repo root

### Error produced:
> /Users/gebert/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_hw_support/libesp_hw_support.a(sleep_modes.c.obj):(.iram1.19.literal+0x60): undefined reference to \`phy_close_rf'
/Users/gebert/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_hw_support/libesp_hw_support.a(sleep_modes.c.obj): in function \`rtc_cntl_ll_ext1_set_wakeup_pins':
/Users/gebert/esp/esp-idf/components/hal/esp32/include/hal/rtc_cntl_ll.h:36: undefined reference to `phy_close_rf'
collect2: error: ld returned 1 exit status
make[3]: *** [jade.elf] Error 1
make[2]: *** [CMakeFiles/jade.elf.dir/all] Error 2
make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
make: *** [flash] Error 2

JamieDriver, 2023-09-12:
I think this may be a bug with `esp-idf v5.0.2`.
The issue appeared when we moved from `5.0.1` to `5.0.2` - I haven't been able to resolve it locally.

I tested bumping the Jade the fw to `esp-idf v5.1.1` and that NORADIO build then succeeds - I flashed an m5stack-fire and all seemed good.

Will look to effect that upgrade asap ...

JamieDriver, 2023-09-25:
`master` branch has been updated to esp-idf version `5.1.1` and this issue appears to be resolved.
Please close this issue if the NORADIO configuration now build for you.
Thanks.

JamieDriver, 2023-10-06:
Closing as believed fixed and no activity.
Please comment/reopen if you still have issues.
Cheers.

closed 2023-10-06 by JamieDriver

GebertJung, 2023-10-06:
Hi Jamie, sorry for getting back to you on this late. Seems all in ordner now on master (7cb104cda) on my end as well. Thanks for your help.