Blockstream/Jade #85: Using `sign_psbt` causes the Jade to crash

otkstudio, 2023-08-24:
Hi,

I'm using the `sign_psbt` endpoint but it's causing the Jade to crash. We're running 2of3 multisig with the config registered successfully on the Jades. Here is the function call for the `sign_psbt` message. The experience is that the Jade immediately turns off and reboots, landing in a locked state.

```
export const signPsbt = async (port: SerialPort, worker: Worker, network: string, psbt: string): Promise<string | void> => {
  const sign_psbt: Message = {
    id: '6979',
    method: 'sign_psbt',
    params: {
      network,
      psbt: bytesFromBase64(psbt),
    },
  }

  console.log("Message:", psbt, sign_psbt)

  try {
    const response = await worker.handleMessage(encode(sign_psbt), port)
    if (isResponse(response) && typeof response.result === 'string') {
      const psbt: string = response.result
      return psbt
    }
  } catch (error) {
    // removed for brevity
  }
}
```

Here is what is logged
```
Message: cHNidP8BAFIBAAAAAS3QiWM1nkpB44dFLIRsOF9jIKPIqxW9Ab+u+7Vwf5LjAAAAAAD9////AVUDAAAAAAAAFgAUh3B3EwIHRWJoDaDd6nYt5Rnxh40ewyUAAAEA6gIAAAAAAQEgN48SVsx6Ln8Xqe7yTBFPV3o1AHUPqIXa8Gh0dQmohQAAAAAA/f///wLoAwAAAAAAACIAIHT7AGgVUhB15sA2hfE+6afcRjqf2Rp3XantE2ssT6eP84iIAAAAAAAWABTieTDf+jlafFb3+5h31A/8xsSZiAJHMEQCIHwhjjxZY4658Y1Wr5+BxAUA2VjHtcqUjBzad7UTcleLAiBXbF/0RGvm9wbvCe7gGHarq6OZaNsMBwcjFSVTFwFv6AEhAjmJb1ndX4fkN8169T77XRvwzBu2Fx7LQ6Z4IXunoGYoDsMlAAEBK+gDAAAAAAAAIgAgdPsAaBVSEHXmwDaF8T7pp9xGOp/ZGnddqe0TayxPp48BBWlSIQM3l5f8CleBlOSmEX6hPYWcJ6pPI4TlM9ECaIYGc8bAOCEDYDnMnL7dp/3Hfsx+cdsugdqGvoTmYfNfGrTPiXj1d38hAxAM9uz362wLVsGyUyqwPxKzLMk1AF96ClKOz2AIRMAAU64iBgMQDPbs9+tsC1bBslMqsD8SsyzJNQBfegpSjs9gCETAABwQTkzqMAAAgAEAAIAAAACAAgAAgAAAAAAAAAAAIgYDN5eX/ApXgZTkphF+oT2FnCeqTyOE5TPRAmiGBnPGwDgMdzesuAAAAAAAAAAAIgYDYDnMnL7dp/3Hfsx+cdsugdqGvoTmYfNfGrTPiXj1d38cJMRrejAAAIABAACAAAAAgAIAAIAAAAAAAAAAAAAA
{
    "id": "6979",
    "method": "sign_psbt",
    "params": {
        "network": "testnet",
        "psbt":  ArrayBuffer(660)
    }
}
```

Here are the two attempted translations from base60 to bytes for the message

```
const bytesFromBase64 = (base64String: string) => {
   return Uint8Array.from(atob(base64String), c => c.charCodeAt(0))
 }

function bytesFromBase64(base64: string) {
  var binary_string = window.atob(base64);
  var len = binary_string.length;
  var bytes = new Uint8Array(len);
  for (var i = 0; i < len; i++) {
      bytes[i] = binary_string.charCodeAt(i);
  }
  return bytes.buffer;
}
```

Hoping this is due to a failed conversion from base64, any help is much appreciated.

JamieDriver, 2023-08-25:
Thanks for all this info - super helpful.
Can reproduce - looks like an issue our side.  ;-(

FYI/notes:
* multisig registration looks good/correct - standard unsorted-multi wsh 2of3.
* psbt looks good/standard, nothing unusual here - one input needing signing by this multisig, one output.
* if I set jade wallet to some other unrelated seed, the call works and returns the PSBT unchanged  (there was nothing for my unrelated signer to do).  But you can see it 'works' on jade etc, so all b64/psbt encoding should be fine.
* if I **DO NOT** register the multisig wallet (or delete it) on jade, the sign psbt call works perfectly with your signer, and returns the psbt updated with that signer's signature.
[ NOTE: jade does not need the multisig registered to sign atm, just to produce/validate receive and change addresses ]

So I think we (jade) have an issue when trying to match input/output utxos to potentially relevant multisig records.
But def our side - I will investigate as a top priority.

otkstudio, 2023-08-25:
Thanks also for the detailed response! Appreciate you investigating it for us and marking it as top priority. I'll roll with unregistered signing for the moment and update when resolved.

moneymanolis, 2023-08-25:
@JamieDriver I am wondering whether the empty array for the derivation path of one of the signers is the culprit

```
        "derivation":[]
```

JamieDriver, 2023-08-25:
I believe I have this fixed locally - just need to get a second set of eyes (PR, review etc.) and it'll get merged asap.

JamieDriver, 2023-08-28:
Merged to master.  Will be included in next fw version.  Will leave item open until tested/agreed resolved.

JamieDriver, 2023-09-20:
Should be resolved in fw `1.0.23` - please close if you agree.
Thanks.

moneymanolis, 2023-09-20:
@JamieDriver can be closed, thanks!

closed 2023-09-20 by JamieDriver