Blockstream/Jade #88: Airgap issues: missing xpub of addresses "1..." + scan addresses failure using other paths

Ciugam, 2023-09-03:
Hi,
I've noticed that using airgapped Jade I can export only "wrapped segwit" and "native segwit" xpubs.
Using the cable instead, I can create wallets using wrapped+native segwit and the oldest legacy addresses (1...).
Could you please add the legacy xpub export in airgapped mode?
I have also another little issue.
With cable I can search for addresses between multiple accounts and paths, no issues.
Unfortunately, using airgapped mode, Jade scans successfully only the standard path addresses (m/44'/0'/0'/0'/n), while other paths are not considered (ex. change addresses or m/44'/0'/0'/m'/n).
Could be added an option, during "Scan QR", asking in which path I want to search for my address?
Many thanks in advance.

JamieDriver, 2023-09-04:
Neither seem unreasonable, cc: @bitcoinhelp ?

JamieDriver, 2023-09-04:
See also #75

JamieDriver, 2023-09-05:
Will address your first point (support of 1... addresses (m/44'/...)) in the next fw release.

Will include your second point in a later release, as will be implemented with #75.  (Both good points and def something we want to include.)
Many thanks for the suggestions.

JamieDriver, 2023-09-20:
First point (support of 1... addresses (m/44'/...)) should vbe addressed in fw `1.0.23`

JamieDriver, 2023-10-06:
FYI Planning to include setting account for xpub export, and account and change flag for address qr scan verification in upcoming fw release `1.0.24`.

Ciugam, 2023-10-22:
Hi, I'm just trying the 1.0.24beta2 and it's simply perfect.
To be curious, how much time before seeing the taproot descriptor?
Thank you in advance

JamieDriver, 2023-10-27:
`1.0.24` released with (I hope) the behaviour requested.
Closing this issue as resolved. 
Please reopen if what's implemented is not what you meant!
Thanks.

closed 2023-10-27 by JamieDriver