Blockstream/Jade #89: Feature request: a time-locked transaction backup service

James-Sangalli, 2023-09-06:
Hi, I was wondering if you were considering adding a time-locked backup service similar to this: https://github.com/James-Sangalli/crypto-timelocked-backup, but for Bitcoin only. 

I recently spoke with Marek from Trezor, and he told me that they are creating such a product in Trezor suite, albeit with a multisig.

Obviously you already support the creation of time-locked transactions, but such a service would require a lot more parts, such as notifying a user when a timelock is close to reaching its date, notifying a user when the time-locked backup has been invalidated, choosing a recipient, prompting the user to create new backups/invalidate old ones, and etc. 

Is this something you have been thinking about?

JamieDriver, 2023-09-08:
This sounds more like something that would be handled by the Green wallet app - Jade could ofc be used to sign the txns, but I think as far as Jade is concerned it would just look like signing any other multisig.
ie. I don't see jade requiring any fw changes to support this idea.

cc: @bitcoinhelp

bitcoinhelp, 2023-09-11:
Nice - yeah as Jamie said this would be more of a Green feature

James-Sangalli, 2023-09-29:
Thank you @bitcoinhelp & @JamieDriver.

Is this a service you would consider supporting from the green app? I.e. an actual time-locked backup service built directly in the app, with you holding the private key of the recipient address, providing alerts to users & so on?

closed 2023-09-29 by James-Sangalli