Blockstream/Jade #90: Cannot reproduce build on MacOS M2

pedromtcosta, 2023-09-11:
I am following the instructions at [REPRODUCIBLE.md](https://github.com/Blockstream/Jade/blob/master/REPRODUCIBLE.md).
I copied the Jade 1.1 BLE configs (`./production/sdkconfig_jade_v1_1_prod.defaults`) and followed the steps until signing the build binary with the 'dev' key.

After that, I downloaded the firmware with the command:
```
pip install -r ./requirements.txt
python ./jade_ota.py --skipserial --skipble --write-compressed --download-firmware --release stable --hw-target jade1.1
```
and selected the version `2)  1.0.21 - ble`. The compressed binary's hash matches the one I see when I initiate the update process from Blockstream Green and the uncompressed binary hash matches the one from the corresponding .hash file.

Now to get the diff of the downloaded and built binaries:
```
xxd build/jade_signed.bin > jade_signed.hex
xxd build/1.0.21_ble_1314816_fw.bin > target.hex
diff jade_signed.hex target.hex
```
This is the diff I get:
[diff.txt](https://github.com/Blockstream/Jade/files/12579755/diff.txt)

It seems to me that this diff contains more changes than only the signature, but maybe I am just interpreting it incorrectly?
Is this expected, or am I doing something wrong on this verification process?

JamieDriver, 2023-09-12:
No that's not expected.
```
< 00060d30: 484f 5354 2d61 6172 6368 3634 2d6c 696e  HOST-aarch64-lin
< 00060d40: 7578 2d67 6e75 2f78 7465 6e73 612d 6573  ux-gnu/xtensa-es
< 00060d50: 7033 322d 656c 662f 7372 632f 6e65 776c  p32-elf/src/newl
< 00060d60: 6962 2f6e 6577 6c69 622f 6c69 6263 2f73  ib/newlib/libc/s
< 00060d70: 7464 6c69 622f 6474 6f61 2e63 0042 616c  tdlib/dtoa.c.Bal
< 00060d80: 6c6f 6320 7375 6363 6565 6465 6400 2f62  loc succeeded./b
< 00060d90: 7569 6c64 732f 6964 662f 6372 6f73 7374  uilds/idf/crosst
< 00060da0: 6f6f 6c2d 4e47 2f2e 6275 696c 642f 484f  ool-NG/.build/HO
< 00060db0: 5354 2d61 6172 6368 3634 2d6c 696e 7578  ST-aarch64-linux
< 00060dc0: 2d67 6e75 2f78 7465 6e73 612d 6573 7033  -gnu/xtensa-esp3
< 00060dd0: 322d 656c 662f 7372 632f 6e65 776c 6962  2-elf/src/newlib
< 00060de0: 2f6e 6577 6c69 622f 6c69 6263 2f73 7464  /newlib/libc/std
< 00060df0: 6c69 622f 6764 746f 612d 6765 7468 6578  lib/gdtoa-gethex
```
vs
```
> 00060d30: 7874 656e 7361 2d65 7370 3332 2d65 6c66  xtensa-esp32-elf
> 00060d40: 2f73 7263 2f6e 6577 6c69 622f 6e65 776c  /src/newlib/newl
> 00060d50: 6962 2f6c 6962 632f 7374 646c 6962 2f64  ib/libc/stdlib/d
> 00060d60: 746f 612e 6300 4261 6c6c 6f63 2073 7563  toa.c.Balloc suc
> 00060d70: 6365 6564 6564 002f 6275 696c 6473 2f69  ceeded./builds/i
> 00060d80: 6466 2f63 726f 7373 746f 6f6c 2d4e 472f  df/crosstool-NG/
> 00060d90: 2e62 7569 6c64 2f78 7465 6e73 612d 6573  .build/xtensa-es
> 00060da0: 7033 322d 656c 662f 7372 632f 6e65 776c  p32-elf/src/newl
> 00060db0: 6962 2f6e 6577 6c69 622f 6c69 6263 2f73  ib/newlib/libc/s
> 00060dc0: 7464 6c69 622f 6764 746f 612d 6765 7468  tdlib/gdtoa-geth
> 00060dd0: 6578 2e63 0000 0000 1e07 423f 1c58 453f  ex.c......B?.XE?
```
One example - It looks like slightly different paths are being encoded.

I think it's fair to say the reproducible build instructions/setup cannot be complete/correct for your platform/architecture.

pedromtcosta, 2023-09-16:
Following up on this issue, I cannot reproduce the build on Ubuntu 23.04 either following the same steps as I described above.

pedromtcosta, 2023-09-16:
This is the diff I get when building on Ubuntu, does it have any clues about what I might be doing wrong?
[diff_ubuntu.txt](https://github.com/Blockstream/Jade/files/12640770/diff_ubuntu.txt)

JamieDriver, 2024-01-11:
Hopefully resolved as per #98