Blockstream/Jade #91: Enhancement: Remove the "repeat delay" when longpressing the the jog-wheel

smartsourcefreak, 2023-09-12:
When longpressing the jog-wheel for example to enter a seed phrase manually it's a bit annoying to have the "repeat delay" when longpressing the wheel to the left or the right to quickly navigate to a certain letter. After the first move/step its annoying to have to wait that long before the cursor moves further.

Would be awesome to have that removed because it would enable a much quicker control, more responsive navigation.

smartsourcefreak, 2023-09-14:
> Obviously its present so what is intended as a single click isn't interpreted as multiple - but we can certainly look to make the delay smaller.

Awesome! Thanks a lot.

smartsourcefreak, 2023-09-18:
Just another idea: Why not have a topic in settings along with screen brightness, where we have "Key repeat delay" with the options "Off, Short Long". That would be really focussing on individual user experience.

JamieDriver, 2023-10-27:
Are you finding that any better in the latest fw release, `1.0.24` ?

The lib that handles button events was updated, and this slightly altered the repeat behaviour.  
(Obviously we can't make it too sensitive, or it double 'events' when you only intended single.)

Yes, the user preference/setting is a fair idea.

smartsourcefreak, 2024-01-07:
Hey. Not noticing an real difference for now. Actually overall it's okay like it is, the only really annoying part where I wish it would be much shorter is, when I recovery a wallet by words and navigate over the "keyboard". The initial delay only during this process could likely be reduced to around 33% or even 25%. Thanks for workin on it, though.