Blockstream/Jade #93: Firmware for ESP32-S3

smartm0use, 2023-09-27:
Are there chances that one day Jade could be flashed on ESP32-S3 board? The one with 1,9 inch display...

[image]

JamieDriver, 2023-09-28:
This could be in the works... albeit as a lower-priority background task ... ;-)  
There are some pre-requisites we need to address first - the display library/code being the main issue.  
It's not top of our list, I'll be honest - but it is something we want to support in the long run.  

Will leave this item open and will update with any progress ...

JamieDriver, 2023-09-28:
NOTE: the button-handler library has been updated in master - which was a prerequisite for using this chip.

JamieDriver, 2023-10-27:
Further supporting changes in master - hopefully we'll have support for this chip in the next release or two.

smartm0use, 2023-10-27:
Great! Keep up the good work :)

FranciscoSa, 2024-01-15:
Do we have any updates on this?

athoszxz, 2024-01-22:
[image]
I believed for a moment and failed miserably.

JamieDriver, 2024-01-23:
No, there's a fair amount of work to get it to work on this chip ... but that work is being done when nothing more urgent takes priority.  Hopefully mid-year ...  
We'll be initially looking at the lilygo-tdisplays3 (as shown above) and the m5stack core-s3 I think.

bobbarbosa, 2024-02-22:
I'm eagerly awaiting the release of the version for "S3". The screen is better and meets a more modern proposal. What doesn't help much is the position of the buttons or the fact that, perhaps, it isn't used vertically.
How about taking advantage of the screen and format and making it vertical?

JamieDriver, 2024-08-05:
We have finally merged the branch adding (still early days / experimental, at this point ...) esp32s3 support.  
tdisplays3 should be supported.  
Have at it ;-)

dark-thirty, 2024-08-14:
I'm testing new esp32s3 firmware and everything seems to be fine.

Custom board - *Waveshare ESP32-S3 DevKit N8R8, OV7670 Camera module and ILI9341 2.8" SPI TFT LCD Display*:

https://github.com/user-attachments/assets/0e86c223-45d8-4058-9c6a-fa3720f9b514

Surprisingly OV7670 camera works with ~25fps. QR code scan works OK. 

I found a little UI bug in Session/QR Scan. 
There is an outlined rectangle at the bottom of the screen, perhaps related to the position of the virtual buttons for the M5 CORE S3:

[image]

ty ^_^

:ok_hand:

JamieDriver, 2024-08-15:
Wow that's awesome!
We concentrated our initial dev/testing on the tdisplays3 and tdisplays3procamera models, so anything else may be a bit shaky atm.  Yes, someone played with the CoreS3 but I don't think it quite works yet - but as you say it does draw buttons on the touch-screen (but that ought to be controlled by `CONFIG_BOARD_TYPE_M5_CORES3` - so not 100% sure).

If it's only on the camera screen it may be an artifact from the camera or the logic that scales the image for display, etc.

Hopefully esp32-s3 will get a bit more love and polish over the next few releases.

dark-thirty, 2024-08-15:
> Wow that's awesome!

Тhank you!

> If it's only on the camera screen it may be an artifact from the camera or the logic that scales the image for display, etc.

This is definitely not a camera artifact,  this bug only appear when i'm in an active session in jade and if i select Scan QR from the menu for example to verify wallet address.
Here is a high quality photo:

[image]

Then if I log out and choose QR Mode, this bug goes away:

[image]

JamieDriver, 2024-08-15:
ahh, I think that's the (atm empty) progress bar that would fill up if you were scanning eg a multi-frame/animated QR, like a large PSBT.

dark-thirty, 2024-08-15:
> ie. it's not a bug, it's a feature! 😆

I see!  :joy:

Keep up the good work.

Sandmann21s, 2024-08-28:
Do you think it would be possible to integrate this camera into a TTGO T-Display S3? This would reduce the overall size of the setup and lower the assembly costs, in addition to giving it a more professional appearance. I would be more than happy to develop a nice 3D case for it, as well as a battery module, just like I've been doing for the Jade DIY, Krux, and Seed Signer hardware wallets.

https://github.com/Blockstream/Jade/issues/93#issuecomment-2288941895

dark-thirty, 2024-08-28:
> Do you think it would be possible to integrate this camera into a TTGO T-Display S3? 

I don't have a T-Display S3 board, but I think there are not enough free pins available.

Sandmann21s, 2024-08-30:
The ttgo s3 is a board that has a lot of available pins and the advantage is that it already comes with a screen, which makes the final assembly more beautiful and I believe more economical, it costs around $15 to $20.
[image]

greenaddress, 2024-09-24:
@smartm0use @bobbarbosa @FranciscoSa @athoszxz @dark-thirty 

master should currently work on some S3 devices. Additionally tdisplays3 and tdisplays3procamera can be flashed directly from https://blockstream.github.io/jadediyflasher/ (source code for DIY flasher is here https://github.com/Blockstream/jadediyflasher )

Do not yse the DIY flasher other than for testing/evaluation purposes. these devices in debug mode aren't really safe and the web page can't setup secure boot for you (at least for now)

Some features are still not working well across S3 devices (like USB storage) but we welcome patches.

bubelov, 2025-05-06:
> Do not yse the DIY flasher other than for testing/evaluation purposes. these devices in debug mode aren't really safe and the web page can't setup secure boot for you (at least for now)

I saw some people on Twitter recommending that flashing tool, it would be nice to add a warning about debug mode being enabled by default and the dangers that it poses

Got an S3 Touch LCD 2 to test, works pretty well with Jade so far

Sami32, 2025-07-28:
Tiny but handy ;-)

https://lilygo.cc/products/t-dongle-s3