Blockstream/Jade #96: Serial connection issue with DIY

andrea-dandrea, 2023-11-03:
Good afternoon,

so I tried the DIY Jade wallet on a M5StickV Plus.
I built version 1.0.23 and flashed the stick.
It apparently works via bluetooth but has issues via usb and doesn't connect.
The Jade wallet reacts when I plug it on the PC (both usb logo and battery power up), but the PC doesn't recognize it any more.
I built version 1.0.24 but I cannot flash if because the serial port doesn't connect/work.
I checked the logs of when I flashed the stick first and it looks that righ after starting up the pc loses the serial port.
At the moment I am not able to bring the stick back to factory status and start all over again.
I guess the factory reset in Jade menu just resets the wallet and not the stick itself.
Any suggestion will be appreciated.

Andrea

nate922, 2024-01-27:
Hi Andrea,.

I believe this issue is due to the USB device ID of the M5StickC.

M5StickC uses VendorID:0403 ProductID:6001 which is different to retail jade.
Sparrow uses HWI which recently has this device ID added.

You can recompile jade with this device ID and it should work.

Update: Can we please add this device id into newer version of Jade?

robertclarkson, 2024-02-27:
I'm having a similar issue with the device not being found in Green wallet using USB with the Wrover Cam ESP32. The device shows up as 
```
Bus 001 Device 009: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
```

JamieDriver, 2024-03-27:
When connecting via USB the app has to scan looking for a (potential) Jade device.  
It is possible that the app in question does not have the vid/pid of your device.  
The list we have (in our internal ci tests) is:
```
    # Used when searching for devices that might be a Jade/compatible hw
    JADE_DEVICE_IDS = [(0x10c4, 0xea60), (0x1a86, 0x55d4), (0x0403, 0x6001), (0x1a86, 0x7523)]
```

mauriciorlima, 2024-07-04:
> Ao conectar via USB, o aplicativo tem que escanear procurando por um dispositivo Jade (potencial). É possível que o aplicativo em questão não tenha o vid/pid do seu dispositivo. A lista que temos (em nossos testes internos de ci) é:
> 
> ```
>     # Used when searching for devices that might be a Jade/compatible hw
>     JADE_DEVICE_IDS = [(0x10c4, 0xea60), (0x1a86, 0x55d4), (0x0403, 0x6001), (0x1a86, 0x7523)]
> ```

Has this error been corrected yet?

JamieDriver, 2024-07-05:
Jade's own list of devices to scan for is:
```
    JADE_DEVICE_IDS = [
            (0x10c4, 0xea60), (0x1a86, 0x55d4), (0x0403, 0x6001),
            (0x1a86, 0x7523), (0x303a, 0x4001), (0x303a, 0x1001)]
```
I believe these are (were) updated to include diy devices when support for that device is added.

If some other wallet software, eg Sparrow, is not using that full list, it indeed won't see some devices - but there is nothing we can directly do from here.  If anything, you'd need to raise an issue with the wallet in question.

After the upcoming `1.0.31` release, when further devices should be supported (eg. M5StickCPlus2 :crossed_fingers: ) I will raise issues against these other projects to update their lists to the master list as in our repo.

JamieDriver, 2024-09-05:
I have recently made a PR to HWI which should address this issue - ofc we have to wait for HWI to take them, release, and then Sparrow to upgrade HWI ...
See https://github.com/bitcoin-core/HWI/pull/753

Sami32, 2025-04-12:
Your HWI PR has already been merged.

Any reason to still keep this issue open?

JamieDriver, 2025-04-14:
Not really.
The support for all devices should percolate through to all HWI-using wallet apps as they each update their HWI version.
Closing.
Thanks.

closed 2025-04-14 by JamieDriver