Blockstream/Jade #98: Non-determinism in builds?

Giszmo, 2023-11-08:
Checking the reproducibility of binaries I occasionally get a build that does not match. So the build succeeds but is different than the release.

Out of 10 compilations that happens not more than twice but it does happen.

* [compileJade.log](https://github.com/Blockstream/Jade/files/13300201/compileJade.log)
* [diffoscopeBuiltStripped.txt](https://github.com/Blockstream/Jade/files/13300213/diffoscopeBuiltStripped.txt)

JamieDriver, 2023-11-10:
Agree - aware - not had time to investigate properly as yet.
Suspecting that some of the dependencies in the docker image are not pinned - unconfirmed.

JamieDriver, 2023-11-10:
tbh the linux deps in the docker image shouldn't make any difference to the esp32 binary.  I mean ofc the esp32 libs and toolchain etc. need to be the same, sure.  But linux deps shouldn't end up in the compiled esp32 binary.  So if the sha of the image is different but the final binaries are the same, I don't think that matters too much in and of itself.

So long as:
a) you can inspect the Dockerfile and build it
b) you can then use the image to build the fw - and the fw ends up the same (sig aside).

I have sync'd my repo to tag 1.0.24 (I thought I'd pick the one prior - more time for the images to diverge) and got two versions of the image - one pulled using the sha in the .gitlab-ci.yml (ie. the exact one used to build the released binaries), and one built 'now' from the Dockerfile in the repo at that tag.
The two images are indeed different.

```
jamie@athens:~/blockstream/jade$ grep "image:" .gitlab-ci.yml 
image: blockstream/verde@sha256:bb20e230564195a3a9f8b4dfa338af15da8c06862ecd487cf3b7e09279435055
jamie@athens:~/blockstream/jade$ docker pull blockstream/verde@sha256:bb20e230564195a3a9f8b4dfa338af15da8c06862ecd487cf3b7e09279435055
...
docker.io/blockstream/verde@sha256:bb20e230564195a3a9f8b4dfa338af15da8c06862ecd487cf3b7e09279435055
jamie@athens:~/blockstream/jade$ docker tag 700db533c5dc blockstream/verde:1.0.24_pulled
...
...
jamie@athens:~/blockstream/jade$ DOCKER_BUILDKIT=1 docker build . -t blockstream/verde:1.0.24_local
...
 => => writing image sha256:e2495370461ee24d240bb612f46cb6bd8e4d1e95229e8662996e73f48cea0d68                  0.0s
 => => naming to docker.io/blockstream/verde:1.0.24_local 
...
jamie@athens:~/blockstream/jade$ docker images --digests
REPOSITORY                                TAG                 DIGEST                                                                    IMAGE ID       CREATED          SIZE
blockstream/verde                         1.0.24_local        <none>                                                                    e2495370461e   22 minutes ago   3.77GB
blockstream/verde                         1.0.24_pulled       sha256:bb20e230564195a3a9f8b4dfa338af15da8c06862ecd487cf3b7e09279435055   700db533c5dc   4 weeks ago      3.76GB
```

However, when I use them to build a binary I get the same output:
```
jamie@athens:~/blockstream/jade$ docker run -v ${PWD}:/jade -it blockstream/verde:1.0.24_pulled bash
root@df632a103431:/jade# cd jade
root@df632a103431:/jade# . ~/esp/esp-idf/export.sh
root@df632a103431:/jade# idf.py -B build.pulled all
...
root@df632a103431:/jade# sha256sum build.pulled/jade.bin
2fce56d60d1206f878eeecbd0a9a6d203e4783f09c2624694e7be41e13cb496b  build.pulled/jade.bin
```

```
jamie@athens:~/blockstream/jade$ docker run -v ${PWD}:/jade -it blockstream/verde:1.0.24_local bash
root@747de43d2362:/jade# cd jade
root@747de43d2362:/jade# . ~/esp/esp-idf/export.sh
root@747de43d2362:/jade# idf.py -B build.local all
...
root@747de43d2362:/jade# sha256sum build.local/jade.bin
2fce56d60d1206f878eeecbd0a9a6d203e4783f09c2624694e7be41e13cb496b  build.local/jade.bin
```

Unfortunately neither of these matched what the actual build produced on the day ~2wks ago.
```
jamie@athens:~/Downloads/build_prod$ ls -lrt
total 2308
-rw-r--r-- 1 jamie jamie    8192 Oct 24 09:22 ota_data_initial.bin
-rw-r--r-- 1 jamie jamie 1376256 Oct 24 09:24 jade.bin
-rw-r--r-- 1 jamie jamie   63418 Oct 24 09:24 sdkconfig
-rw-r--r-- 1 jamie jamie     356 Oct 24 09:24 index.json
-rw-r--r-- 1 jamie jamie    1738 Oct 24 09:24 bsdiff.h
-rw-r--r-- 1 jamie jamie    9697 Oct 24 09:24 bsdiff.c
-rw-r--r-- 1 jamie jamie      64 Oct 24 09:24 1.0.24_ble_1376256_fw.bin.hash
-rw-r--r-- 1 jamie jamie  880293 Oct 24 09:24 1.0.24_ble_1376256_fw.bin
drwx------ 2 jamie jamie    4096 Nov 10 17:17 bootloader
drwx------ 2 jamie jamie    4096 Nov 10 17:17 partition_table
jamie@athens:~/Downloads/build_prod$ sha256sum jade.bin
20d1ab2b5afba78e6db6be9595dba366432e4f87ed0e6347be5a24082e96e2f6  jade.bin
```

Then I read your '8 out of 10' comment more closely, and decided to rebuild from clean N times in each container.
In each container, *sometimes* I get `2fce...496b`, and *sometimes* I get (the expected) `20d1...e2f6` !
Non-determinism indeed!

With your builds - was it inside the same container/image ?  ie. do you agree you can get differences rebuilding inside the same container ? 

Ok, I can capture all the build outputs and intermediate files and hopefully find out exactly which components are building in this non-deterministic way !!!
Thanks for the pointer!

JamieDriver, 2023-11-10:
OK, in all cases the sha hashes of all the object/intermediate files are identical - so the difference is something to do with (the order of?) linking.
Ongoing ...

JamieDriver, 2023-11-13:
I tracked this down in the end to a memory segment issue with the underlying espressif esp-idf tools which generate a linker input file.
I raised an issue with espressif - and it transpires that they have fixed this issue recently (but it is not yet in a tagged release).  See: https://github.com/espressif/esp-idf/issues/12573.

We should pick up this fix when they next release and we track - if this looks like it's going to be a long time we may consider patching our idf tools, but we'd rather avoid that and keep a 'clean' dependency tree if possible.

Hopefully it'll be fixed in the next Jade release or so - we'll keep this item open until then ofc.

JamieDriver, 2023-12-05:
fyi this fix has now been released upstream and the dependency updated in our `master` branch, so should be fixed as of the next jade fw release.

JamieDriver, 2024-01-11:
Should be resolved as of fw `v1.0.27`

Giszmo, 2024-03-08:
Sorry for the late reply. I just tried out to reproduce the still latest fw release 1.0.27 and found only one flavor - `10_ble` - to be reproducible.

My full report:

```
$ scripts/test/hardware/blockstreamjade.sh 1.0.27
...
Results:
db3ea043b891c862401d110923be04b0c2ef9b100c57cad22ca345b032efc448  jade_1.0.27_10_ble.bin
60b78e2b4f04ed982e00002ee54235e87c39128f9dbab0529025ee4eb9950716  jade_1.0.27_10_ble.built.bin
60b78e2b4f04ed982e00002ee54235e87c39128f9dbab0529025ee4eb9950716  jade_1.0.27_10_ble_stripped.bin
The Jade firmware version 1.0.27 10_ble is reproducible with above hashes.
b2b67f5943ca4c1ae0c7fc3733352dbb4b644add5a174111547e5bd7f77d7171  jade_1.0.27_10_noR.bin
b75b61cb3952382e3fe3b8ccd20e94c2bec4cca165fb29e2d82aad24d2bfbd26  jade_1.0.27_10_noR.built.bin
ec1ef3e8890d9b6b3be8d7ad2b8261f605e4d6071910073124a595f95f764cbb  jade_1.0.27_10_noR_stripped.bin
The Jade firmware version 1.0.27 10_noR is **not** reproducible with above hashes.
9cd8a3abc492499468cb4a7bbdd70d20063ca670bf0d0485f7044c1f494c0723  jade_1.0.27_11_ble.bin
9322fa18ee83f21afe8b9c1c758f49d007d19a340332238d4439e8261e641232  jade_1.0.27_11_ble.built.bin
3e3569b48248140782be55354e1470a12b2e52d10ff4891953ebee743a96f093  jade_1.0.27_11_ble_stripped.bin
The Jade firmware version 1.0.27 11_ble is **not** reproducible with above hashes.
cd5c9439037b1c1b94eb93fc774d146e24918f60cd61cac9bc8c008e80adc9fe  jade_1.0.27_11_noR.bin
c4f46f32b91c03fcdd72989a75f036ddb6adc8282a3293ccb6e91f01b53a050c  jade_1.0.27_11_noR.built.bin
470b1f32f788feb021ac07180c96e9d7553aaa0748a6e1cda969bb9220564564  jade_1.0.27_11_noR_stripped.bin
The Jade firmware version 1.0.27 11_noR is **not** reproducible with above hashes.
```

Only the firmware in the `10_ble` variant was reproducible. The others not.

A closer look at the diff:

```
$ diffoscope /tmp/jade/jade_1.0.27_10_noR{.built,_stripped}.bin                                            |    0%                             ETA:  --:--:-- 
--- /tmp/jade/jade_1.0.27_10_noR.built.bin
+++ /tmp/jade/jade_1.0.27_10_noR_stripped.bin
@@ -1,20 +1,20 @@
 00000000: e907 0220 3815 0840 ee00 0000 0000 032c  ... 8..@.......,
 00000010: 018f 0100 0000 0001 2000 403f 14e8 0500  ........ .@?....
 00000020: 3254 cdab 0000 0000 0000 0000 0000 0000  2T..............
-00000030: 312e 302e 3237 2d64 6972 7479 0000 0000  1.0.27-dirty....
+00000030: 312e 302e 3237 0000 0000 0000 0000 0000  1.0.27..........
 00000040: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000050: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000060: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000070: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000080: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000090: 7635 2e31 2e32 0000 0000 0000 0000 0000  v5.1.2..........
 000000a0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-000000b0: f895 ffcc 9a0a 6b78 83d5 f6c7 4903 bd51  ......kx....I..Q
-000000c0: f485 e171 571d 1c1b 46a4 badc 19c2 b0ee  ...qW...F.......
+000000b0: 303e 885a d829 62ab c756 539e ad41 3834  0>.Z.)b..VS..A84
+000000c0: 850a d806 6ef0 483b 1b3e 4dbd e1b3 bcd6  ....n.H;.>M.....
 000000d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 000000e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 000000f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000100: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000110: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000120: 014a 4144 4500 0000 0000 0000 0000 0000  .JADE...........
 00000130: 0000 0000 0000 0000 0000 0000 0000 0000  ................
@@ -69623,10 +69623,10 @@
 0010ff60: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0010ff70: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0010ff80: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0010ff90: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0010ffa0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0010ffb0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0010ffc0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-0010ffd0: 0000 0000 0000 0000 0000 0000 0000 002d  ...............-
-0010ffe0: 232f 4de0 9daa a0f4 cf25 b6ef 45ca c2ae  #/M......%..E...
-0010fff0: 0f1a c7e4 5d6c f9af 91ca b192 24a1 da92  ....]l......$...
+0010ffd0: 0000 0000 0000 0000 0000 0000 0000 00ca  ................
[... 62 lines cut ...]
+000000c0: 55e7 3f69 b5e8 5f90 6b3f 264b 31de ce54  U.?i.._.k?&K1..T
 000000d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 000000e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 000000f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000100: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000110: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000120: 014a 4144 455f 5631 2e31 0000 0000 0000  .JADE_V1.1......
 00000130: 0000 0000 0000 0000 0000 0000 0000 0000  ................
@@ -90103,10 +90103,10 @@
 0015ff60: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0015ff70: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0015ff80: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0015ff90: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0015ffa0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 0015ffb0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
[... cut ...]
-0015ffd0: 0000 0000 0000 0000 0000 0000 0000 00b0  ................
-0015ffe0: ca94 3197 a5fe edc5 b6d3 8bd4 e785 e4b6  ..1.............
-0015fff0: 673f 87f6 f44f 9840 cf85 fc0d 3b74 2381  g?...O.@....;t#.
+0015ffd0: 0000 0000 0000 0000 0000 0000 0000 00b4  ................
+0015ffe0: bd31 7fc4 c48d 11c0 dd52 078b 8dc7 7228  .1.......R....r(
+0015fff0: bf5e dd06 d833 6ecc ed80 5312 1089 bfba  .^...3n...S.....
```

While this looks like a tiny diff, we can't comment on this being benign or not.

JamieDriver, 2024-03-11:
```
-00000030: 312e 302e 3237 2d64 6972 7479 0000 0000  1.0.27-dirty....
+00000030: 312e 302e 3237 0000 0000 0000 0000 0000  1.0.27..........
```
This suggests that the repo being built is not 'clean' - ie. has been changed from a simple 'git fetch'.

Can you provide the output of 'git diff' and/or 'git status' on the repo dir ?
Many Thanks.

Giszmo, 2024-03-11:
Good points. I'll try to fix my script accordingly.

In `git status` I see files I downloaded and compiled for example but in `git diff` I see this, which I don't immediately know how to fix.

```
diff --git a/dependencies.lock b/dependencies.lock
index 5a7796f..84b7669 100644
--- a/dependencies.lock
+++ b/dependencies.lock
@@ -40,6 +40,6 @@ dependencies:
     source:
       type: idf
     version: 5.1.2
-manifest_hash: dcb6c640489c3318dcfbc90652ff7cb8af8391868cf3c5e86a46ad26c39b42f0
+manifest_hash: b234b600e781b408de1324c4e0733ab400773e7502a418c911bb03a55bc35dfb
 target: esp32
 version: 1.0.0
```

Giszmo, 2024-03-11:
Ok, replacing the "git diff" with a "git stash" fixed the reproducibility. I will update our review asap.

closed 2024-03-11 by Giszmo

JamieDriver, 2024-03-11:
Really appreciate your time and diligence here.  Many, many thanks.  :+1: