Blockstream/Jade #99: Can't recover wallet on verson 1.0.25

dgcasana, 2023-11-09:
I try jade Diy on TTGO T-Display and when I try to restore my wallet via seed, keyboard appears but no cursor and device seems freeze.
Only occurs on version 1.0.25 (can't try 1.0.26).
On commit 1.0.24 all works ok.

JamieDriver, 2023-11-09:
Yes, regression/bug in 1.0.25 - 1.0.26 was made available (almost immediately) to address this issue.
1.0.24 and 1.0.26 should both be fine - 1.0.25 should now be unavailable for download from the official channels.

NOTE: 1.0.26 was made by reverting the offending commits on branch `release_1.0.26` - master is still broken in this regard as the 'proper fix' (as opposed to a simple revert) is being tested/reviewed.

Many thanks.

JamieDriver, 2023-11-13:
FYI: this should now be resolved in master.

closed 2024-01-11 by JamieDriver