Blockstream/Jade #101: Screen Flickering

MyNameIsOka, 2023-11-25:
This is probably not something that can be fixed but I would like to know if that happens often before I buy another Jade.

I bought one of the first batches of the Jade but haven't been using it that much. I have been playing around with it recently and I noticed that the screen suddenly goes dark quite frequently, when clicking the button or scrolling.

Initially I thought the device died but after a few moments the display comes back up again. Sometimes it's dark for quite some time, though.

https://github.com/Blockstream/Jade/assets/18796117/e512362a-3701-48d7-9cbe-0e7707ec58d1

Is that a known problem? Is that still occurring in the newer models?

MyNameIsOka, 2023-12-04:
Thank you for your response. Yes, that's how it looks to me, too. I just wonder if that is something that is occuring often?

JamieDriver, 2023-12-14:
I am not aware ... @bitcoinhelp ?

bitcoinhelp, 2023-12-14:
Very strange, no that's not something I've seen before... If you want to reach out to help.blockstream.com we can look into a resolution here

MyNameIsOka, 2023-12-15:
All good, thanks. I ordered a new unit a few days ago. 
As I said, I pre-ordered the current device when it came out, so it was one of the earlier batches and maybe had some shortcomings.

closed 2023-12-22 by MyNameIsOka