Blockstream/Jade #106: Feature request: taproot support with export relative xpub

Ciugam, 2023-12-13:
Hello!
As the title says, it could be possible to add taproot support?
This descriptor was added 2 years ago, and I would try to sign txs using this new address.
Also, it could be fine to add xpub export for taproot.
Thank you in advance

MotivaoCrypto, 2024-01-08:
Hi @Ciugam , I would also like the possibility of Taproot Support (BIP86)

Kokufuu, 2024-04-25:
I would also like to see and use that feature. +1

Drew72-ita, 2024-05-13:
Yay me as well ^_^

LukasErdmanski, 2024-07-03:
I would alsko like to see and use it.

Tacitoku, 2024-10-08:
Although [Taproot adoption](https://txstats.com/d/000000055/taproot-statistics?orgId=1) is currently ~0.46% and [P2WPKH](https://txstats.com/d/000000016/p2wpkh-statistics?orgId=1) is ~28%, there are some privacy features in Taproot that make it very attractive despite its low adoption. I am curious whether there exists a roadmap when Taproot will be implemented on Jade.

Sami32, 2024-12-09:
https://github.com/Blockstream/green_android/issues/233#issuecomment-2506809249

https://github.com/Blockstream/gdk/tree/taproot

mamolas, 2025-01-22:
P2TR (Pay-to-Taproot) multisig is superior because it:
Has better privacy - all multisig transactions look like single-sig on-chain
Reduces transaction size and fees
Provides script flexibility through Tapscript

+10 to this feature