Blockstream/Jade #107: Issue with using Jade in multi sig with Sparrow Wallet

wealthcollector, 2023-12-19:
Hi guys,

Just got a new Jade wallet and I'm using it in multi sig with sparrow wallet. I currently have 5 UTXOS in this wallet. Earlier when I tried to sign with my jade for the 2 smallest UTXOS in the wallet, everything worked fine. However, when I tried to sign a transaction on 3 bigger UTXOS, it says processing for a long time then on sparrow it keeps showing "read nothing, apparent EOF", while other UTXOS say "failed to extract input_tx" or  "messaging/communication error".  Now when I try, all 5 UTXOS won't work, so it's not a small/large UTXO issue. The other hardware wallets in the multi sig are working fine and signing but can't get jade to work on these specific UTXOS. I've input the passcode and its "active" but just can't get it to sign a transaction. Hopefully someone can help me out. Not sure if this is a sparrow issue, jade issue, or specific jade issue. I think it's a specific jade issue cause I have another jade and it works just fine with a different multi sig set up.

Thanks.

wealthcollector, 2023-12-19:
After I click "sign" on sparrow the "output 1/1" screen pops up the jade. When I press the arrow to confirm, it just goes back to the main screen where is says "session". It doesn't go to the next screen  where it says "send transaction" with the fee, like it does when i try to sign for the smaller transactions

dextersl2, 2024-01-04:
Anyone using a 
Blockstream Jade in the stateless mode in conjunction with SparrowWallet and getting an error when signing a transaction? Error: Failed to extract input_tx

wealthcollector, 2024-01-04:
Are you using a Mac? Downloading the driver made it work for me. @dextersl2 

https://help.blockstream.com/hc/en-us/articles/900005443223-Fix-issues-connecting-Jade-via-USB#h_01H4PFV1R8CC9TCSBBMBTP5E6J

https://github.com/sparrowwallet/sparrow/issues/1203

https://github.com/sparrowwallet/sparrow/issues/1060

JamieDriver, 2024-01-11:
@dextersl2 - This means sparrow is not passing the utxo tx - is this the latest version of sparrow ?

JamieDriver, 2024-09-05:
I have recently made a PR to HWI which should address this - ofc we have to wait for HWI to take them, release, and then Sparrow to upgrade HWI ...
See https://github.com/bitcoin-core/HWI/pull/753