Blockstream/Jade #109: Secret wallet

106Madara, 2023-12-29:
"I wanted to create a second wallet with a passphrase and use it as my main wallet. The issue I see is that this second wallet, where I have the additional passphrase, is still visible on the interface. I want this second 'secret' wallet to be invisible, so that in case of any situation, no one can see that I have a second wallet. It would be great if there was a way to make a wallet invisible."

JamieDriver, 2024-01-11:
Sorry, not sure what you mean.
You can use a recovery-phrase to set up Jade, either scanning/entering every time (stateless) or storing it encrypted on Jade and protecting that with a PIN.
If you want to, you can further use a passphrase to access separate wallets.

eg.
* put a small amount of funds in the passphrase-less wallet as a 'dummy'
* use a passphrase to access your larger wallet, containing more of your funds

I'm not sure what you mean by 'visible on the interface' ?

Preferences can either be set to either a) require a passphrase every time you enter the PIN, or b) not ask for a passphrase (assume no/null passphrase).  [ In the latter case you would need to go to 'Options->BIP39 Passphrase->Once' before attempting to unlock Jade to tell it that on this one occasion you want to enter a passphrase ... ]

106Madara, 2024-01-11:
Sure, here's the English version of your message:

Yes, I understand. It would be a good feature if the wallet with a passphrase were not visible during login. In the unfortunate event of a robbery, the thief would still see two wallets – one with fewer satoshis and one with more satoshis. They would notice that I have two wallets, and it would be beneficial if the second wallet with a passphrase were not visible when I plug in the Jade and try to log in. In that case, the thief won't see that I have more satoshis on the wallet protected only by a code, without a passphrase. In other words, he sees only one wallet and doesn't realize that I have an additional wallet with a passphrase.

JamieDriver, 2024-01-12:
It sounds like you're talking about the Green app, rather than Jade.  @bitcoinhelp ?

106Madara, 2024-01-12:
Yes, exactly. My mistake; I wrote in the wrong Git... I noticed it after responding to your comment. I'd like to apologize for wasting your time. I'm new to GitHub sorry.

I already wrote the same thing to the correct Git. I hope it's the right Git this time. 
I apologize once again.

JamieDriver, 2024-01-15:
No problem at all !
Feedback is much appreciated.
Cheers.

JamieDriver, 2024-03-27:
Closing here as issue for companion app rather than Jade iteself.  
Many thanks for the useful discussion!

closed 2024-03-27 by JamieDriver