Blockstream/Jade #112: Unusual change path warning

Zeke8990, 2024-01-11:
I didn’t have any issues sending/receiving from other wallets to Jade. However, I’m having trouble consolidating UTXOs within my Jade, using Sparrow.

When I attempt to sign the transaction, Jade warns me of an “unusual change path” and Sparrow reports that it “failed to extract input _tx.”

I have closed Jade in Sparrow and reopened with the correct passphrase, but don’t know what to try next.

JamieDriver, 2024-01-12:
> it “failed to extract input _tx.”
This means sparrow is not passing the utxo tx - is this the latest version of sparrow ?
Seems v odd that it would work for a foreign address but not for your own address - I assume that's the only difference ?
You're doing a 'send all' to an address of your own ?

georgantas, 2024-07-27:
Related to: https://github.com/Blockstream/Jade/issues/40

The jade does not handle consolidation transactions properly.

JamieDriver, 2024-07-29:
In what sense ?

The 'unusual path' warning is more of a 'mismatch' between Jade and the app in question - ie. if the app tells Jade that a certain output is our 'change' - but the path is not of the internal chain (ie. is not .../1/n) then Jade issues that warning - same as if the app tells Jade that a certain output is *not* our change (but is eg a consolidation output back to ourselves) but the path *is* of the internal chain (ie. .../1/n) then again Jade issues that warning.

The error “failed to extract input_tx.” is due to the app not passing Jade the full tx from which the utxo being spent was an output.

georgantas, 2024-07-29:
As I understand it, the mismatch is due to a bug in HWI, which most apps use to interface with Jade. I don’t foresee app developers fixing the issue in HWI any time soon.

JamieDriver, 2024-07-29:
I can look at it myself when I get a moment ...

georgantas, 2024-07-29:
Do you know if resolving the mismatch in HWI would also fix the second issue (`failed to extract input_tx`) ?

JamieDriver, 2024-07-30:
I don't think so, as I think that implies the psbt being passed from sparrow to hwi/jade is missing some data jade expects.

CaryCreative, 2024-08-17:
I am having the same issue, error messages as above when using Sparrow with the Jade to consolidate UTXOs  , is there a solution. Any help would be appreciated.

georgantas, 2024-08-28:
I'm not aware of any workaround at the moment.

CaryCreative, 2024-08-28:
Hi Yianni,Thanks for the update. Please keep me posted if there's a fix for this issue. Currently, I can't move funds from Sparrow Wallet due to problems with bulk transactions from Strike, though transactions work from Green Wallet.In Sparrow, the UTXO appeared normal but couldn't be signed because of an error. Additionally, Jade incorrectly reports that the fee is larger than the transaction amount. But the fee is actually significantly lower. The main issue:- Jade displays a warning about an "unusual change path."- Sparrow fails with a "failed to extract input_tx" error.Appreciate any help. Thanks!Cary Hammond Sent on the move

georgantas, 2024-08-28:
Oh actually, you can maybe try using an older version of Sparrow. I don’t recall “failed to extract input_tx” being an issue when I had used multisig with the Jade some versions ago. I believe I had used this version: https://github.com/sparrowwallet/sparrow/releases/tag/1.6.4

The unusual change path error will persist however because thats an issue on the Jade’s side.

CaryCreative, 2024-08-29:
Hi, Yianni Thank you for the advice I shall try to revert back to an older version of Sparrow. Maybe that might work. The issue was with one UTXO from Strike which was sent when I was using the latest Sparrow update. I’ll try an older version and send a transaction from strike and see if I get the same signing issue. I’m using Green to & that seems fine just in case. Interesting on the unusual change path maybe there will be a fix on a future firmware update. Was the other issue too where Jade shows a message saying that the fee is higher than the send if that makes sense. Which it’s not but I still sign out a transaction fine. Thank you Yianni for your time with this. The Jade is a great little device but you do get confused - concerned when things don’t seem correct 👍 Cary Hammond Sent on the move

georgantas, 2024-08-31:
No, I did not get any fee issues on that version. If you are still having issues with the fee and "invalid input" error on version 1.6.4 of Sparrow, then I think these issues could be related to a newer version of the firmware. I don't think a root cause for them has been identified yet. Agree, these issues do not inspire confidence 😣

JamieDriver, 2024-09-05:
The 'unusual path' warning is more of a 'mismatch' between Jade and the app in question - ie. if the app tells Jade that a certain output is our 'change' - but the path is not of the internal chain (ie. is not .../1/n) then Jade issues that warning - same as if the app tells Jade that a certain output is not our change (but is eg a consolidation output back to ourselves) but the path is of the internal chain (ie. .../1/n) then again Jade issues that warning.

The error “failed to extract input_tx.” is due to the app not passing Jade the full tx from which the utxo being spent was an output.

JamieDriver, 2024-09-05:
I have recently made a PR to HWI which should address these issues - ofc we have to wait for HWI to take them, release, and then Sparrow to upgrade HWI ... 
See https://github.com/bitcoin-core/HWI/pull/753

CaryCreative, 2024-09-05:
Hi Jamie very Interesting and thank you for your response. I also want to thank the team at Blockstream, I have had some excellent feedback. 

‘Change’ popped up when I was trying to research the issue I was having but did not quite understand at the time. 

I also spoke with Craig Raw on Telegramme Funny as at first it was not him and I was given instructions for a fix, well step 2 was to input my seed which of course I did not LOL 

Anyway, I did speak with the real man who advised on a driver issue but that did not work. He then advised me to speak with Blookstream. 

So Jamie this transaction was a single UTXO from Strike as a bulk send, I have since stopped until I can solve this issue with Sparrow as it was fine when using green. 

What would be the advice now Jamie would I just have to wait for Sparrow to upgrade HWI, which I have no idea what that is ha! 

Unfortunately the link https://github.com/bitcoin-core/HWI/pull/753. I get a warning from and can't see that page. 

Thanks again