Blockstream/Jade #113: possible to deactivate everthing except camera

skyerjoe, 2024-01-23:
Is it possobe to deactivate ( likely on a hardrware layer) everthing except qr camera on jade ?

JamieDriver, 2024-01-24:
There is a firmware build which does not include any of the BLE/radio drivers - it's not disabled on the chip, we just omit the drivers from the fw package.  
Is this sufficient/the sort of thing you mean ?  

atm there is no equivalent for USB - but making such a fw change would be possible.

I'm not sure it'd really be desirable though - if you did that you'd never be able to update the fw on the jade if security patches, important fixes, or new functionality you are interested in was released.

Also atm if you have the full fw including ble/radio drivers, you can still go into a menu and disable/unload the drivers.  In order to access that menu and switch it to re-enable/load the drivers, you need to know the PIN of the device (assuming you are using a PIN-protected wallet).  We could perhaps consider something like that for USB - would that suffice ?

skyerjoe, 2024-02-10:
Is there any adantages if u deacivate this on the hardware layer from ur point of view ? 
Is this possible ? 
Btw. Could u pint out the link with fw version ? 
Best regards

skyerjoe, 2024-02-14:
Thnx for ur help 
best regards

closed 2024-02-14 by skyerjoe