Blockstream/Jade #115: Support for M5StickC PLUS2

nate922, 2024-01-25:
Currently we have support for M5StickC and it works great.

This is being phased out for the PLUS2 Version with a different display pinout and Secure Boot V2.

The current Jade firmware results in a black screen.

greenaddress, 2024-01-26:
There are quite a few changes, we could speculate changes but until we get one we can't really do too much.
It has a new power chip( axp 192 vs new) and different pinout. If anyone wants to provide a PR it would be welcome, otherwise we hope to get one relatively soon and add support but it isn't top priority right now

tortxoFFoxtrot, 2024-02-20:
Sad

SKALOVSK, 2024-03-22:
Is there any way to install it? I bought the M5 and only now did I see that it is not compatible

viniciusmoraissilva, 2024-05-21:
please add a way to support the M5StickC PLUS2, as the normal m5 stick will probably soon be discontinued, and new hardware has more possibilities and more longevity and more possibilities.
Soon users will probably want to migrate to the M5StickC PLUS2.

nate922, 2024-05-21:
@3rdIteration is the man when it comes to M5StickC support.

I believe he was playing with one Plus2 version so hopefully there's something in the pipeline.

He has a github here if you want to follow:
https://github.com/3rdIteration/Jade

viniciusmoraissilva, 2024-05-21:
If I doubt it until they port it to the M5StickC PLUS2 I'm already old and decrepit.

3rdIteration, 2024-05-21:
I have one here that I'm add a config profile for once I have finished the Trezor diy video. (Which will be out this week)

nate922, 2024-05-21:
Thanks @3rdIteration  - Look forward to the diy trezor video and love your work.

viniciusmoraissilva, 2024-05-21:
👀 for this video thank you bro.

YuriCPS, 2024-05-29:
> I have one here that I'm add a config profile for once I have finished the Trezor diy video. (Which will be out this week)

Wow, great to read this. I love watching videos on YouTube and I just watched the trezor video, I liked example 2 with the small display. But I liked Jade more than Trezor.

Thank you very much for your contribution.

3rdIteration, 2024-06-04:
Basically there are a few tweaks required in terms of how it handles power, etc, so it's not something that adding a new sdkconfig in this instance. 

That said, the changes require are straightforward. For those who have something they want to test, I have added initial support in this form here: https://github.com/3rdIteration/Jade/tree/M5StickCPlus2

I haven't tested things like the calibration of the battery level indicator or voltage, haven't tested whether it sleeps properly or not, but it's a good starting point and I'll fix up the other stuff over the coming days.

On the hardware side, it's the same as the original M5StickC-Plus in most ways, though the larger battery and the extra button is nice. The other thing to add is that the Plus2 change the SoC such that it also now supports Secure Boot V2, so that's nice too.

Once it's all done I'll bring a PR for the main repo.

3rdIteration, 2024-06-04:
Fixed up the battery calibration and sleep, everything should be working as expected now :)

viniciusmoraissilva, 2024-06-05:
@3rdIteration I am having a problem, my friend. Simply put, files are missing both when cloning the directory and when downloading it manually. These files are only from the folders inside the jade>components directory.

To clone the directory, I am using: git clone -b M5StickCPlus2 https://github.com/3rdIteration/Jade.git.

But since that didn't work, I downloaded the zip and placed it manually, and still, the same files from the folders inside jade>components are missing.

Am I doing something wrong or is it a bug? I need your help, thank you.

3rdIteration, 2024-06-05:
What operating system are you on? (Just in terms of how you are using Git)

viniciusmoraissilva, 2024-06-05:
@3rdIteration: windows brow

3rdIteration, 2024-06-05:
Simplest solution on windows is to use GitHub desktop, then just you download my repo (it will do it recursively by default) then just switch the branch.

viniciusmoraissilva, 2024-06-05:
I actually haven't tried it through the repository using GitHub Desktop, I will try. Thanks, man.

nate922, 2024-06-05:
Working perfectly. Second button is pretty cool. Really appreciate your work @3rdIteration 

BTW is the secureboot V2 good to go yet? Any pointers so I don't screw it up.

I did V1 on the older stick, I take it it's similar to that?

viniciusmoraissilva, 2024-06-05:
Dude, I loved it. Simple and intuitive, the M5 Stick C Plus 2 is wonderful for Jade. I thought I'd go crazy if I didn't see this support. @3rdIteration, you deserve a kiss on the neck. Man, thank you, you are a friend.

nate922, 2024-06-06:
Secure boot V2 works great. Unfortunately it doesn't connect to Blockstream green due to the USB ID so I couldn't verify.

Only issue is that the git clone from your repo reports the firmware version as 3a44322 (commit number)
Sparrow doesn't like this as a SemVer string. Is there anyway to change this on the code ?

Thanks @3rdIteration 

(2 parents [68b40f1](https://github.com/Blockstream/Jade/commit/68b40f12d22941d0b33f6ec3265dd53469007697) + [3de931a](https://github.com/Blockstream/Jade/commit/3de931aec2b7430b66b043dce3bf31f398bd50b3)
commit 3a44322)

3rdIteration, 2024-06-06:
I'll check the USB device IDs and may need to make some additional PRs so that the Jade drivers in various software packages pick it up. (It works nicely over bluetooth with a phone at the moment)

I also just added a release tag, so that should make the version string stuff behave better.

3rdIteration, 2024-06-06:
One thing to add is that it seem to be getting a network or server error when connecting via USB. (Though it is correctly detected by things like Blockstream Green, etc) 

I'll have a look at what is causing this over the next little while. (Though it does work just fine over bluetooth)

3rdIteration, 2024-06-06:
Fixed in latest push. (It just shows USB as being connected all the time now, will see if there is a better way to identify this with the hardware here)

nate922, 2024-06-06:
Thanks, flashed on the Plus2 stick, seems to be working ok, I will try it out today.

I'm having some issues on the Plus1 stick, it had secureboot v1 enabled, I upgraded to  3a44322.
Now I can't update to your latest version, seems to be ok up to the firmware verification screen. I hit ok and the device reboots, debug level logs don't show anything (I tried a factory reset)

Could this be related to the USB issues? Any pointers on what I could do here? Thanks.

---

Received msg: {'id': '479995', 'result': {'JADE_VERSION': '3a44322', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'M5STICKCPLUS', 'JADE_FEATURES': 'SB', 'IDF_VERSION': 'v5.1.4', 'CHIP_FEATURES': '33000000', 'EFUSEMAC': 'D4D4DA9822F8', 'BATTERY_STATUS': 5, 'JADE_STATE': 'UNINIT', 'JADE_NETWORKS': 'ALL', 'JADE_HAS_PIN': False}}
Running OTA on: {'JADE_VERSION': '3a44322', 'JADE_OTA_MAX_CHUNK': 4096, 'JADE_CONFIG': 'BLE', 'BOARD_TYPE': 'M5STICKCPLUS', 'JADE_FEATURES': 'SB', 'IDF_VERSION': 'v5.1.4', 'CHIP_FEATURES': '33000000', 'EFUSEMAC': 'D4D4DA9822F8', 'BATTERY_STATUS': 5, 'JADE_STATE': 'UNINIT', 'JADE_NETWORKS': 'ALL', 'JADE_HAS_PIN': False}
Sending: {'method': 'ota', 'id': '695857', 'params': {'fwsize': 1245172, 'cmpsize': 810257, 'cmphash': '20e82cc13608e9d8b0872eee5b8968196b4b5abcd5e97d506193dd3cbe874642', 'fwhash': '78ec0b97669f1a989712808319b19cec75173423c8c909d7e16a382c5c7963c3'}} as cbor of size 138
Sending: 138 bytes
Sent: 138 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 2 bytes...
Received: 2 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Received msg: {'id': '695857', 'result': True}
Sending ota_data message 551985 as cbor of size 4133
Sending: 4133 bytes
Sent: 4133 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 2 bytes...
Received: 2 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Received msg: {'id': '551985', 'result': True}
3122.99 b/s - progress 0.51% - 258.14 seconds left
Written 4096b in 1.31s
Sending ota_data message 462232 as cbor of size 4133
Sending: 4133 bytes
Sent: 4133 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 2 bytes...
Received: 2 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Reading 6 bytes...
Received: 6 bytes
Reading 1 bytes...
Received: 1 bytes
Received msg: {'id': '462232', 'result': True}
10004.57 b/s - progress 1.01% - 168.50 seconds left
Written 8192b in 1.72s
Sending ota_data message 891986 as cbor of size 4133
Sending: 4133 bytes
Sent: 4133 bytes
Reading 1 bytes...

3rdIteration, 2024-06-06:
I have heard of someone else who had a similar issue, but I haven't been able to reproduce. I'll give it a go with a plus1 I have here over the next few days.

nate922, 2024-06-07:
Thank you. BTW New firmware works great over serial on desktop (sparrow/green)

3rdIteration, 2024-06-17:
https://github.com/Blockstream/Jade/pull/141

JamieDriver, 2024-06-18:
Love it man, this is awesome!  
We're in the middle of a bit of work in this area internally, so if it conflicts a lot it might have to wait for a couple of weeks for the dust to settle with some merges etc before we can look to fold this in - but will definitely circle back to it asap.  
Many thanks, this is very much appreciated!

tortxoFFoxtrot, 2024-06-18:
So good with the Bitcoin Orange theme 👌🏻

Now I want to work out how to code in XMR keys 😬

[image]

tortxoFFoxtrot, 2024-06-18:
[image]

SKALOVSK, 2024-06-21:
[image]
Would anyone know how to solve this problem?

tortxoFFoxtrot, 2024-06-21:
> ![image](https://private-user-images.githubusercontent.com/164506194/341611947-1aaa0465-31fc-40a1-812f-a78c096e3319.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Did you try what it is telling you to do in the meassage?

JamieDriver, 2024-06-21:
The Jade repo has submodules - so either:
a) when you clone the repo, use `git clone --recursive ...`  
b) if you used vanilla/'shallow' clone, you can fetch the submodules using `git submodule update --recursive --init`

You should then be able to see `esp32_deflate` under the `components` directory:
```
jade$ ls -l components/
total 32
drwxr-xr-x 2 jamie jamie 4096 Feb 14 15:05 assets
drwxr-xr-x 2 jamie jamie 4096 Dec 15  2023 autogenlang
drwxr-xr-x 5 jamie jamie 4096 May 22 08:18 esp32_bc-ur
drwxr-xr-x 3 jamie jamie 4096 Mar 25  2022 esp32_bsdiff
drwxr-xr-x 2 jamie jamie 4096 Dec 12  2023 esp32_deflate
drwxr-xr-x 5 jamie jamie 4096 Nov  6  2023 esp32-quirc
drwxr-xr-x 3 jamie jamie 4096 Dec 15  2023 esp32-rotary-encoder
drwxr-xr-x 4 jamie jamie 4096 Feb 14 15:05 libwally-core

jade$ ls -l components/esp32_deflate/
total 28
-rw-r--r-- 1 jamie jamie  124 May 24  2022 CMakeLists.txt
-rw-r--r-- 1 jamie jamie 7375 Dec 12  2023 deflate.c
-rw-r--r-- 1 jamie jamie 3881 Dec 12  2023 deflate.h
-rw-r--r-- 1 jamie jamie  501 May 24  2022 Kconfig.projbuild
-rw-r--r-- 1 jamie jamie 1086 May 24  2022 LICENSE
-rw-r--r-- 1 jamie jamie  459 May 24  2022 README.md
```

It should then build :crossed_fingers:

JamieDriver, 2024-08-09:
M5StackCPlus2 support to master @d48d60f3 - many thanks to @3rdIteration

closed 2024-08-09 by JamieDriver

tortxoFFoxtrot, 2024-08-10:
I got my Mt Gox bitcoin into Kraken so i will move it onto my jade haha